nfc-tools/libnfc #458: No NFC device found when used ACR1252u

ehsansis0, 2017-10-16:
hi

i want to work on nfc. but when i run nfc-list command return No NFC device found.

i check with version 1.5, 1.5.1 and 1.7 but still doesn't work.

shibinsundar, 2017-11-16:
Me too have the same issue. ACR1252U is not detecting...
**# nfc-list
nfc-list uses libnfc 1.7.1
No NFC device found.**

thican, 2017-11-16:
Hello everyone,

I confirm I have this same issue, and was working on it a long time ago already.
I recently made a post on the GNU/Linux distribution I use, to provide more information on the steps I followed : [[Gentoo] [NFC] Unable to get nfc-scan-device working unlike pcsc_scan](https://forums.gentoo.org/viewtopic-p-8138738.html)

Thanks for support.

Matheeus97, 2019-03-06:
Any update about this problem?

graynk, 2019-06-13:
> Any update about this problem?

I don't know if this will work for ACR1252u, since I have ACR122u, but I did the following:
1. Recompiled libnfc with ./configure

graynk, 2019-06-13:
> Any update about this problem?

I don't know if this will work for ACR1252u, since I have ACR122u, but I did the following:
1. Recompile libnfc with `./configure --with-drivers=acr122_usb`, then `sudo make install`
2. `echo -e 'blacklist nfc\nblacklist pn533\nblacklist pn533_usb' > /etc/modprobe.d/blacklist-libnfc.conf`
At this point nfc-list should work under root. But I found it troublesome, so I continued:
3. Find vendorId and productId of your reader in `lsusb`
4. `sudo groupadd usb`
5. `sudo usermod -a -G usb $USER`
6. `echo 'SUBSYSTEMS=="usb", ATTRS{idVendor}=="YOUR_VENDOR_ID", ATTRS{idProduct}=="YOUR_PRODUCT_ID", SYMLINK+="sub20", GROUP="usb", MODE="660"' > /etc/udev/rules.d/99-libnfc.rules`
7. `sudo udevadm trigger`
Now nfc-list will work without root. Depending on your setup you might even be able to remove pcsc-tools, the wrapper over libnfc that I use sadly requires it, but nfc-list itself works without it.

stigok, 2019-08-06:
Just to mention that `sudo echo "text" > /etc/udev/file.rules` will not work, as the redirection itself is not run with sudo, just `echo`. Use `echo 'SUBSYSTEMS=="usb", ATTRS{idVendor}=="YOUR_VENDOR_ID", ATTRS{idProduct}=="YOUR_PRODUCT_ID", SYMLINK+="sub20", GROUP="usb", MODE="660"' | sudo tee -a /etc/udev/rules.d/99-libnfc.rules` instead.

rheaalleen, 2019-10-24:
Tried @graynk solution but the reader is still missing from nfc-list, pcsc_scan works fine.

Hasnt much use for me tho since mfoc/mfcuk depend on libnfc to work and the ACR1252U isnt recognized

torrunes, 2021-01-15:
This is some time ago now. Is this libnfc support for 1252U still not present?

smortex, 2021-01-15:
ACR1252u is only mentioned in this issue, so we can safely consider it to be unsupported.

closed 2021-01-15 by smortex