nfc-tools/libnfc #465: RPI PN532 Breakout error: libnfc.driver.pn532_spi Unable to wait for SPI data. (RX)

siliconchris1973, 2017-11-06:
Hello All,

I've tried so many combinations now, but nothing works. I am using a Raspberry PI B (latest board revision) and a clean installed Raspbian Jessie updated to the latest versions. My PN532 is a Sunfounder PN532 MODULE FOR RASPBERRY PI - as sun founder so famously claims.

I downloaded libnfc from GitHub, configured and installed it. When I now Issue a nfc-list, I get the below error:

```
nfc-list uses libnfc 1.7.1
error	libnfc.driver.pn532_spi	Unable to wait for SPI data. (RX)
error	libnfc.driver.pn532_spi	Unable to wait for SPI data. (RX)
error	libnfc.driver.pn532_spi	Unable to wait for SPI data. (RX)
pn53x_check_communication: Timeout
error	libnfc.driver.pn532_spi	Unable to wait for SPI data. (RX)
nfc-list: ERROR: Unable to open NFC device: pn532_spi:/dev/spidev0.0:500000

```

I double and triple checked cabling and also the dip switches on the board and everything is fine. I made sure SPI module is loaded (ldmd shows it) and also the devices under /dev are present:

```
lsmod:
spi_bcm2835             7596  0

ls -la /dev/spi*
crw-rw---- 1 root spi 153, 0 Jan  1  1970 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Jan  1  1970 /dev/spidev0.1

```

I also tried the manual installation from libnfc site and the dip-package for debian jessie on RPI

I have not the slightest idea, what could be the cause. 

Can anyone help me?

SloCompTech, 2018-02-13:
Hi,
I also have problems with library on raspberry pi, but I can read tags.
- check if device is correctly connected to raspberry pi (check pinout,connections ...)
- check that switches are in correct position for spi
- try with more suitable speeds for spi https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md (not sure if this will help)

Cheers

667bdrm, 2018-06-08:
Same issue. SPI-only RC522 module (http://aliexpress.com/item/RFID-module-RC522-Kits-S50-13-56-Mhz-6cm-With-Tags-SPI-Write-Read-for-arduino/32565667225.html). Raspberry Pi 3 model B, Raspbian Jessie with latest update. 

Connection:
3.3v - RC522 3.3v
GND - RC522 GND
SPI0 MOSI (PIN 19) - RC522 MOSI
SPI0 MISO (PIN 21) - RC522 MISO
SPI0 SCLK (PIN 23) - RC522 SCK
SD0 DAT1 (PIN 22) - RC522 RST
SPI0 CE0 (PIN 24) - RC522 SDA

/boot/config.txt:
dtparam=spi=on
dtoverlay=spi0-hw-cs

Also tried dtoverlay=spi0-cs dtoverlay=pi3-disable-bt

# ./spidev_test -D /dev/spidev0.0
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

0D 00 00 00 00 00 
00 60 00 00 00 00 
95 FF FF FF FF FF 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
FF DE AD BE EF BA 
AD F0 

# nfc-list -v
nfc-list uses libnfc libnfc-1.7.1-191-g216145f
error	libnfc.driver.pn532_spi	Unable to wait for SPI data. (RX)
error	libnfc.driver.pn532_spi	Unable to wait for SPI data. (RX)
error	libnfc.driver.pn532_spi	Unable to wait for SPI data. (RX)
pn53x_check_communication: Timeout
error	libnfc.driver.pn532_spi	Unable to wait for SPI data. (RX)
nfc-list: ERROR: Unable to open NFC device: pn532_spi:/dev/spidev0.0:500000

Also tried to use different speed.

If I use python MFRC522 as described at http://www.instructables.com/id/RFID-RC522-Raspberry-Pi/ using the same hardware configuration, the read operation succeed:

# python Read.py 
Welcome to the MFRC522 data read example
Press Ctrl-C to stop.
Card detected
Card read UID: 144,0,42,197
Size: 8
Sector 8 [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

Looks like something wrong with pn532_spi driver

SloCompTech, 2018-06-11:
I have same problems with my PN532 occasionally

ChristopheBelpaire, 2018-07-31:
According to [this article](http://wiki.sunfounder.cc/index.php?title=PN532_NFC_Module_for_Raspberry_Pi#SPI_Communication_Instructions_for_Raspberry_Pi) you should use the value pn532_spi:/dev/spidev0.0:50000 as device with a Raspberry pi 3, instead of pn532_spi:/dev/spidev0.0:5000000.
It works for me but it is not stable ...

Schlumpf9, 2020-09-16:
I've got the same problem using PN532 on Raspberry Pi 4 (newest Raspbian Version)...

Even with the changed baudrate I receive:
`nfc-list uses libnfc 1.7.1
error   libnfc.driver.pn532_spi Unable to wait for SPI data. (RX)
pn53x_check_communication: Timeout
error   libnfc.driver.pn532_spi Unable to wait for SPI data. (RX)
nfc-list: ERROR: Unable to open NFC device: pn532_spi:/dev/spidev0.0:5000`

I have also tried the LibNFC v.1.8.0, but exact the same error occurs. With Python I have no problem receiving the UID of a NFC-Tag using SPI therefore the connection to the Waveshare PN532 NFC HAT board is working... Anyone knows how to fix this?

bananakaba, 2020-11-20:
Solution for the Waveshare PN532 NFC HAT (couldn't try others):

The HAT uses D4 (BCM) as chip selection pin. LIBNFC uses CE0.
So connecting those pins should let ur pi communicate with ur HAT. But be careful to check that D4 (BCM) is set to Mode "IN" because if it is set to "OUT" u could damage ur pi.