nfc-tools/libnfc #466: Support for native LinuxNFC

zonque, 2017-11-17:
Bear with me if this is a dumb question, but I'd like to understand on which layer support for plain LinuxNFC (netlink/AF_NFC, ...) would be implemented.

zonque, 2017-11-26:
IOW: is there any real user-space user of LinuxNFC yet?

henri62, 2019-05-06:
That is a good question, I have a similar issue to hook it it to a UART in an embedded board. I asked a question on stackexchange and got really ZERO answers, not even a comment in the last 1/2 year.
(This post: https://unix.stackexchange.com/questions/481917/how-to-get-the-cr95hf-nfc-reader-to-work-in-the-linux-nfc-stack)
I also looked at a lot of places to get (real) answers and there are none findable. No documentation on the linux stack, nothing, except a some powerpoint presentations where a lot of details are missing.
One of these is this one: https://elinux.org/images/a/a9/Elce11_venancio_ortiz.pdf
Most of them are fairly old (>5 y) and there are no example drivers that can looked at to see how to implement a new one. Let alone an application on user level how to test it.

I found some patch sets and stuff for the cr95hf around and tried those, all efforts were a complete dead end. Mostly because you don't know how to trigger the drivers to load/attach them and debates in the community about using deprecated LDISC stuff in the uart drivers (but they don't how to solve it).

So did you ever find out how to solve it? Or find some descent documentation on the linux NFC ?

Now I'm looking at how much work it is when porting libnfc, but I think it is mayor! But the pro is that at least there are tools around the lib so I can test things.

Another option could be to drop the st95hf all together and choose for a supported PN53* variant chip in libnfc. Its a matter of how must time/money you want to spend on getting it to work and what is the cheapest solution at the end.

zonque, 2019-05-06:
The project I had been working for this topic on was given up on, so there was no progress on my side. I'll leave the issue open for further comments though.

JuniorJPDJ, 2020-03-08:
Have anything with nfc in linux world changed?
I'm trying to do like.. anything? with phone and it's bcm2079x chip, but I don't even know where to look at.
I found linux-nfc and this lib, but both seem to be dead.

JuniorJPDJ, 2020-03-08:
https://www.kernel.org/doc/Documentation/networking/nfc.txt
Docs here

JuniorJPDJ, 2020-03-08:
@henri62 https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/nfc/st95hf?h=v5.6-rc4 your device seems to be supported

smortex, 2021-01-04:
Closing since the point of libnfc is that it also work on non-Linux OSes :wink:

closed 2021-01-04 by smortex

dogtopus, 2023-02-28:
@smortex This doesn't make the point invalid, especially considering that a lot of old NFC related tools only work with libnfc, libnfc is getting lagged behind for interface and chips support, and that libnfc has been breaking the Linux NFC stack for years. Adding Linux NFC as another backend for the libnfc API would potentially benefit a lot of people.