nfc-tools/libnfc #467: iso14443b2sr , not all card support uid

kokOa123, 2017-11-26:
on nfc-list I can get all iso14443b2sr tag (uid).

the problem is there is some tag that i not get there uid , that because the common iso14443b2sr tag can get uid by send 0x0b , but there  are some tag that , to get there uid need to read first blocks.

I want to change it on the code and pull requests
but i not found the code that ask the uid from iso14443b2sr by send 0x0b command

@doegox  
thank you all

doegox, 2017-11-26:
See https://github.com/nfc-tools/libnfc/blob/master/libnfc/nfc-internal.c#L186

kokOa123, 2017-11-26:
@doegox  thank you
this need to be lot of code changes becuase to get uid i need to read 4 blocks , this is not only 1 command to run...

by the way , i offer to add to nfc.c function that return(/fill) nfc_target []  tu return all the tag thar connected,
like nfc_initiator_list_passive_targets but  run on all the types

doegox, 2017-11-26:
See https://github.com/nfc-tools/libnfc/blob/master/libnfc/chips/pn53x.c#L1091
There is already more than the initial 0x0b and it should be easier to add your code here too.