nfc-tools/libnfc #473: [Error] nfc/nfc.h: No such file or directory

ElixG, 2018-02-01:
Hi guys i just started working with libnfc and the compiler gives me this error while trying to compile "[Error] nfc/nfc.h: No such file or directory", can you guys help me? did i fail libnfc installation?

aminmrp, 2018-02-15:
@ElixG  Hi , when do you see this error ? At what stage? If you see this error when you test the code `./configure `, use this code :  `autoreconf -Wall -vis`  then use `./configure`

SloCompTech, 2018-02-15:
@ElixG Hi,
I installed libnfc with folowing commands
```
git clone https://github.com/nfc-tools/libnfc.git
cd libnfc
autoreconf -vis
./configure --prefix=/usr --sysconfdir=/etc
make
sudo make install
sudo cp libnfc.conf.sample /etc/nfc/libnfc.conf # Here I added my reader
```
Compling should work with `-lnfc` flag. Else you can add manualy include dir with `-I <incdir>`.

Cheers

neomilium, 2021-01-04:
Thanks @SloCompTech for reply!

closed 2021-01-04 by neomilium