nfc-tools/libnfc #474: No NFC device found

aminmrp, 2018-02-04:
Hi , when i connect NFC device to pc (NXP PN532 V3) and Trying to found it with libnfc 1.7.1 , i see this problem ,
************************************
[hadi@Hadi libnfc-libnfc-1.7.1]$ nfc-list
nfc-list uses libnfc 1.7.1
No NFC device found.
************************************
I used two devices for testing. But my problem was not resolved. What should I do?

dud225, 2018-02-14:
Hello

Could you try 
`
sudo modprobe -r pn533_usb pn533 nfc
`

aminmrp, 2018-02-14:
@dud225 I tested this code.There was no result.The problem is, I do not know where the problem is

jeanthom, 2018-02-15:
I'm not sure I could help a lot, but what is the result of the `lsusb` command on your PC (with the NFC dongle plugged in) ?

aminmrp, 2018-02-16:
@jeanthom this is the result:
 `[hadi@Hadi libnfc-libnfc-1.7.1]$ lsusb
Bus 002 Device 004: ID 04f3:0235 Elan Microelectronics Corp. 
Bus 002 Device 013: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

geosphere, 2019-02-14:
Some issue with NXP PR533 (3.60) and  libnfc 1.7.1

neomilium, 2021-01-04:
Please read carefully the README file.

closed 2021-01-04 by neomilium