nfc-tools/libnfc #475: Unable to claim USB interface (Device or resource busy) Acr122

C1ta, 2018-02-14:
Writing nfc-list answers me with this error
`error libnfc.driver.acr122.usb Unable to claim USB interface (Device or resource busy)`

How i fix this problem? Thanks

stawiski, 2018-11-25:
Did you build nfc-list as root?
Try sudo.

nospam2000, 2018-11-25:
> Device or resource busy

This error message normally means, that some other daemon is already using the device. The command fuser -fu /dev/YOUR_PORT can help you detect such a daemon.

On my Mac i have the problem, that it has a daemon which claims the acr122, so it cannot be used by libnfc. If you have the same problem, then you can disable the daemon.

AdamLaurie, 2018-11-26:
$ cat /etc/modprobe.d/blacklist-libnfc.conf
blacklist nfc
blacklist pn533

closed 2021-01-04 by doegox