nfc-tools/libnfc #477: nfc-emulate-tag crash pn532

edmundhei, 2018-03-08:
When I use nfc-emulate-tag to emulate a tag and read the tag with my galaxy note 3, the note3 **did** able to read the tag **successfully** but the programme exit with error and crash the pn532 which is really confused me.

```
pi@raspberrypi:~/libnfc-1.7.1 $ nfc-emulate-tag 
nfc-emulate-tag uses libnfc 1.7.1
NFC device: pn532_i2c:/dev/i2c-1 opened
nfc-emulate-tag will emulate this ISO14443-A tag:
ISO/IEC 14443A (undefined baud rate) target:
    ATQA (SENS_RES): 00  04  
* UID size: single
* bit frame anticollision supported
       UID (NFCID3): 08  ab  cd  ef  
* Random UID
      SAK (SEL_RES): 09  
* Not compliant with ISO/IEC 14443-4
* Not compliant with ISO/IEC 18092

Fingerprinting based on MIFARE type Identification Procedure:
* MIFARE Mini 0.3K
* SmartMX with MIFARE 1K emulation
Other possible matches based on ATQA & SAK values:
NFC device (configured as target) is now emulating the tag, please touch it with a second NFC device (initiator)
nfc_target_init: Success
nfc_target_emulate_tag: Success
error	libnfc.bus.i2c	Error: wrote only -1 bytes (10 expected).
error	libnfc.driver.pn532_i2c	Unable to transmit data. (TX)
```

 Unless i unplug it otherwise it is not useable

.
```
pi@raspberrypi:~/libnfc-1.7.1 $ nfc-emulate-tag 
nfc-emulate-tag uses libnfc 1.7.1
error	libnfc.bus.i2c	Error: wrote only -1 bytes (10 expected).
error	libnfc.driver.pn532_i2c	Unable to transmit data. (TX)
pn53x_check_communication: Input / Output Error
error	libnfc.bus.i2c	Error: wrote only -1 bytes (10 expected).
error	libnfc.driver.pn532_i2c	Unable to transmit data. (TX)
nfc-emulate-tag: ERROR: Unable to open NFC device
```

>

chelinho139, 2018-06-15:
@edmundhei +1 any updates on this? thanks

Norbz, 2019-05-22:
Same thing here...

lmckoko-ericsson, 2019-07-04:
Same error, does anyone have an update?

actually, the error also happens when running command nfc-list, at the end of the printout we see 

error	libnfc.bus.i2c	Error: wrote only -1 bytes (19 expected).

Thanks!

Jupiops, 2019-12-21:
Same error here on Raspi 4 connecting the pn532 board using I2C.
[Connecting the board over SPI](http://wiki.sunfounder.cc/index.php?title=PN532_NFC_Module_for_Raspberry_Pi#SPI_Communication_Instructions_for_Raspberry_Pi) solved the problem for me, but if I read the emulated Tag with my Android phone the tag has allways the UID 00::00 
[image]

andresmihkelson, 2022-01-09:
I also ran into this using PN532 over I2C on a Raspberry Pi Zero W.