nfc-tools/libnfc #487: Set SAK not from Block0 value

tokamac, 2018-04-27:
Some  tag reader manufacturers do not follow the Mifare spec and use tags with a different value for SAK (e.g. **0**8) and the byte value written in block 0 (e.g. 0x**88**)

For all UID/SAK modifiable Chinese tags, the SAK value is taken from the block 0 byte value , e.g. XXXXXXXXXX88 written in block 0 will automatically set SAK at 88, but XXXXXXXXXX08 in block 0 will return SAK 08.

Currently it is not possible to have a SAK value (the one reported with nfc-list) different from the byte of block0. Because if you manually modify the value in block 0 in the dump file and copy it back to the tag, the SAK value is modified too.

This problem has already been exposed two years ago by @0ut0fb0unds in issue #336 but incorrectly answered as being a simple question about various tag behaviors, some having a read-only SAK and others a R/W SAK, and consequently the issue was closed. This is a mistake, as it's a common issue with various known brands that should trigger an upgrade feature in libnfc.

See for example comment [2189127](https://gist.github.com/alphazo/3303282#gistcomment-2189127) from @tinue

The solution is to be able to modify the SAK while keeping a different value in block 0, where the SAK is usually set from.

This behavior is something that can be achieved using a Proxmark3, so it should be doable with libnfc too.

nospam2000, 2018-06-24:
> For **all** UID/SAK modifiable Chinese tags, the SAK value is taken from the block 0 byte value

This is wrong. My UID writable cards always return SAK as 0x08 even if byte 5 in block 0 is set to 0x88.
As the comment from quantum-x in #336 says: "There are many variations of badges floating around" and yours are different from mine.

> This behavior is something that can be achieved using a Proxmark3

Can you please quote where this information comes from and what specific command is used to change the SAK to a value independent from block0/byte5?

  Michael

eXspir3, 2018-12-12:
Something interesting to Note:

My Chinese Magic Cards do send 0x08 SAK even when changed to 0x88, but the same tag - when give SAK 0x89 will send 0x89 and not 0x09.

RaJiska, 2019-12-07:
Any news on this ? Is there a way of emulating the behavior of the original tag by returning 08h when 88h is actually written to it ?

jimkid202, 2020-02-16:
Agree -did anyone solve this? My clone was perfect except for the 08/88 error. When I change the UID to 08 it flows through to Block 0. Perhaps it's just a limitation of the infineon card?