nfc-tools/libnfc #493: error libnfc.driver.pn532_i2c Application level error detected  (127)

AProzis, 2018-05-23:
Hello all.

I am using Libnfc on a Raspberry pi 3b+ with a PN532 adafruit.
I am trying to read the tag data with the apdu_example.c. When I run the program, I get a error   error	libnfc.driver.pn532_i2c	Application level error detected  (127).
I already try to use SPI communication but no success.
Could you help me please?

Thank you!

arandintday, 2021-05-14:
I got this error too, and why no one replied you?

Zarinia, 2021-08-31:
you shoud copy /etc/nfc/libnfc.conf  to /usr/local/etc/nfc/libnfc.conf
and test with this command
LIBNFC_LOG_LEVEL=3 nfc-list

ScottyDoesKnow, 2023-10-14:
I was confused by this error message, but then realized it was still working (I looked at the linked "No NFC device found" issue above and saw they had the same error but it still worked). If you encounter this, put a card/fob/etc... on the reader and call nfc-list again to see if it works anyways.