nfc-tools/libnfc #494: Read/Write ISO14443B-2 ST SRx card

neomh, 2018-06-08:
Hello

I'm able to detect the card. I can read the UID but noting else.

Any ideas ?

Thank you

lodi-g, 2018-06-27:
See #436

doudz, 2020-02-11:
@lodi-g Could you explain what you mean ?

> You must first init an B tag, and then you can use sub-b types like B2SR for SRx.

lodi-g, 2020-02-11:
> @lodi-g Could you explain what you mean ?
> 
> > You must first init an B tag, and then you can use sub-b types like B2SR for SRx.

It's been a while, but here's a sample that was working for a B2SR tag.

```c
static bool is_tag_present(void) {
    nfc_target ant[1];
    nfc_modulation nm = {.nmt = NMT_ISO14443B, .nbr = NBR_106};
    nfc_initiator_list_passive_targets(pnd, nm, ant, 1);

    nfc_target ant2[1];
    nfc_modulation nm2 = {.nmt = NMT_ISO14443B2SR, .nbr = NBR_106};
    int res = nfc_initiator_list_passive_targets(pnd, nm2, ant2, 1);

    return res == 1;
}
```

You can read more in issue #439