nfc-tools/libnfc #498: ./configure Missing/Outdated Documentation

gudenau, 2018-08-02:
The configure script appears to be missing, not only that but the documentation seems to be out of date. The README also references INSTALL, which does not exist either.

This is the case with the latest commit (216145fba522a8eaefc5c751b8c10198376bc066) as well as tag 1.7.1 (2d4543673e9b76c02679ca8b89259659f1afd932).

daixtrose, 2018-08-28:
@gudenau I just joined the libnfc maintainers and I plan to fix the CMake files in order to work even as submodule. The configure system which is set up in parallel seems to use autoconf, so maybe [this hint](http://www.delorie.com/gnu/docs/autoconf/autoconf_13.html) might help.

Could you please try it out and if it works maybe provide some additional text for the Readme via PR?

sphelan123, 2018-09-06:
@gunennau, I had the same problem here. 
Here's what I found (this is from a NOOB).

once you    git clone https://github.com/nfc-tools/libnfc.git
cd libnfc 
autoreconf -fiv
./configure --prefix=/usr --sysconfdir=/etc
make clean   <-- Optional unless you have old .o files you want to rebuild.
make
At this point doing cd examples/ 
./nfc-poll works fine.
 then I did 
sudo make install   and now when I try to run the one in /usr/bin by typing nfc-poll I get this.

> nfc-poll uses libnfc 1.7.1
> NFC reader: pn532_i2c:/dev/i2c-1 opened
> NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
> nfc_initiator_poll_target: Invalid argument(s)

So, still working on that.

daixtrose, 2018-09-16:
@gudenau can you confirm that @sphelan123's solution worked for you?

gudenau, 2018-09-17:
Sorry, been busy. Will try and test this tomorrow.

gudenau, 2018-09-18:
```
configure: error: The libusb headers are missing
```

```
$ sudo apt-get install libusb-1.0-0-dev 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-1.0-0-dev is already the newest version (2:1.0.21-1).
0 upgraded, 0 newly installed, 0 to remove and 10 not upgraded.
```

How odd.

doegox, 2018-09-18:
you need libusb-dev, libnfc is still using older libusb 0.1

gudenau, 2018-09-18:
That is kind of silly, maybe I can try updating it when I have time.

gudenau, 2018-09-18:
Seems to work.

gudenau, 2018-09-18:
Made #505 to address this.

doegox, 2018-09-18:
the README.md is meant to be understood in the context of a release (https://github.com/nfc-tools/libnfc/releases).
In a release, there is no need to run autoconf and INSTALL exists.
Now if you don't want to use a release but hack around with the latest development sources, please read HACKING.md.
make_release.sh can help you understanding the differences.

closed 2018-09-18 by doegox

gudenau, 2018-09-18:
First I have heard about that.

Oh well.