nfc-tools/libnfc #499: Is this project dead?

daixtrose, 2018-08-10:
Looking at the unresolved issues and the number of unprocessed pull requests, one might get the impression that the original authors discontinued this project. 

In addition the [link to the forums](http://forums.nfc-tools.org/) yields a PHP error message.

- [ ] Are you looking for a new maintainer? 
- [ ] Are there alternative libraries you recommend?

doegox, 2018-08-10:
Hi Markus
We, historical maintainers, don't have much time anymore indeed.
If someone is up to the task, he's welcome :) Are you interested ?
There are no alternative libraries AFAIK.

doegox, 2018-08-10:
The forum is indeed dead and hosted on an infrastructure we don't have access to, from an even older maintainer.

daixtrose, 2018-08-11:
@doegox Thanks for this insight. Since I am no NFC expert I am unsure whether I am the right one to maintain this project, but at least I could address build issues (CMake) and merge pull requests that seem reasonable enough or have enough votes. 

So if you want to add me as collaborator with write access, please go ahead and do so. I would then also propose to add a call for maintainers to README.md

WRT to the forum: is it possible for you to contact the older maintainer in order to obtain the database? Maybe we can resurrect the trove of information therein.

doegox, 2018-08-28:
Hi @daixtrose 
Sorry, back from holidays today.
Meanwhile we discussed and approved to add you as collaborator :)
I added you, maybe you've to visit https://github.com/nfc-tools/libnfc/invitations to finalize it.
We're still digging into the forum thing.

daixtrose, 2018-08-28:
Many thanks @doegox. I hope I can cope with the given responsibility.

closed 2018-09-04 by neomilium