nfc-tools/libnfc #507: pn532 the red one No NFC device found

leninhat, 2018-10-04:
HI, i've been using the libnfc-1.7.1 on ubuntu 18.04 and trying to use the pn532, but not the adafruit breakboard, the one i have is the red one. 
I don´t know if libnfc does not have the support for this device, however this is what appears when i use lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0bda:b009 Realtek Semiconductor Corp. 
Bus 001 Device 004: ID 04f2:b5db Chicony Electronics Co., Ltd 
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 014: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 001 Device 013: ID 2341:0001 Arduino SA Uno (CDC ACM)
Bus 001 Device 007: ID 89e5:101b  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

this is my libnfc.conf.sample
allow_autoscan = false
device.name = "PN532 NFC Board"
device.connstring = "pn532_uart:/dev/ttyUSB0:115200"
i've changed cause the original didn't work  

when i use autoreconf this is what appears
 autom4te: cannot open autom4te.cache/requests: Permission denied
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
and when i use nfc-lis
this is what appears 
nfc-list uses libnfc libnfc-1.7.1
No NFC device found.
I used the device on arduino and worked fine.

akuplensky, 2018-10-10:
the same here. nfc532 and usb-uart devices are fine, wiring is ok. checked them on windows.

akuplensky, 2018-10-10:
`debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	1 device(s) defined by user
debug	libnfc.general	  #0 name: "red_one", connstring: "pn532_uart:/dev/ttyUSB0:115200"
nfc-list uses libnfc libnfc-1.7.1-204-g4ae4cc8
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyUSB0 at 230400 baud.
debug	libnfc.bus.uart	Serial port speed requested to be set to 230400 baud.
debug	libnfc.chip.pn53x	Diagnose
debug	libnfc.chip.pn53x	Timeout value: 500
debug	libnfc.bus.uart	TX: 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
debug	libnfc.chip.pn53x	SAMConfiguration
debug	libnfc.chip.pn53x	Timeout value: 1000
debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 14 01 17 00 
debug	libnfc.bus.uart	Timeout!
debug	libnfc.driver.pn532_uart	Unable to read ACK
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyS3 at 230400 baud.
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyS2 at 230400 baud.
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyS1 at 230400 baud.
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyS0 at 230400 baud.
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.driver.pn532_uart	Attempt to open: /dev/ttyUSB0 at 115200 baud.
debug	libnfc.bus.uart	Serial port speed requested to be set to 115200 baud.
debug	libnfc.chip.pn53x	Diagnose
debug	libnfc.chip.pn53x	Timeout value: 500
debug	libnfc.bus.uart	TX: 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
debug	libnfc.chip.pn53x	SAMConfiguration
debug	libnfc.chip.pn53x	Timeout value: 1000
debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 14 01 17 00 
debug	libnfc.bus.uart	Timeout!
debug	libnfc.driver.pn532_uart	Unable to read ACK
error	libnfc.driver.pn532_uart	pn53x_check_communication error
debug	libnfc.chip.pn53x	InRelease
debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 52 00 da 00 
debug	libnfc.bus.uart	Timeout!
debug	libnfc.driver.pn532_uart	Unable to read ACK
debug	libnfc.general	Unable to open "pn532_uart:/dev/ttyUSB0:115200".
nfc-list: ERROR: Unable to open NFC device: pn532_uart:/dev/ttyUSB0:115200`

akuplensky, 2018-10-10:
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	1 device(s) defined by user
debug	libnfc.general	  #0 name: "red_one", connstring: "pn532_uart:/dev/ttyUSB0:115200"
nfc-list uses libnfc libnfc-1.7.1-204-g4ae4cc8
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyUSB0 at 230400 baud.
debug	libnfc.bus.uart	Serial port speed requested to be set to 230400 baud.
debug	libnfc.chip.pn53x	Diagnose
debug	libnfc.chip.pn53x	Timeout value: 500
debug	libnfc.bus.uart	TX: 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
debug	libnfc.chip.pn53x	SAMConfiguration
debug	libnfc.chip.pn53x	Timeout value: 1000
debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 14 01 17 00 
debug	libnfc.bus.uart	Timeout!
debug	libnfc.driver.pn532_uart	Unable to read ACK
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyS3 at 230400 baud.
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyS2 at 230400 baud.
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyS1 at 230400 baud.
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyS0 at 230400 baud.
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.driver.pn532_uart	Attempt to open: /dev/ttyUSB0 at 115200 baud.
debug	libnfc.bus.uart	Serial port speed requested to be set to 115200 baud.
debug	libnfc.chip.pn53x	Diagnose
debug	libnfc.chip.pn53x	Timeout value: 500
debug	libnfc.bus.uart	TX: 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
debug	libnfc.chip.pn53x	SAMConfiguration
debug	libnfc.chip.pn53x	Timeout value: 1000
debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 14 01 17 00 
debug	libnfc.bus.uart	Timeout!
debug	libnfc.driver.pn532_uart	Unable to read ACK
error	libnfc.driver.pn532_uart	pn53x_check_communication error
debug	libnfc.chip.pn53x	InRelease
debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 52 00 da 00 
debug	libnfc.bus.uart	Timeout!
debug	libnfc.driver.pn532_uart	Unable to read ACK
debug	libnfc.general	Unable to open "pn532_uart:/dev/ttyUSB0:115200".
nfc-list: ERROR: Unable to open NFC device: pn532_uart:/dev/ttyUSB0:115200

neomilium, 2018-10-14:
Log output says there is no reply from device you point.

What is the "red" PN532 ?

akuplensky, 2018-10-15:
@neomilium  
[this one](https://www.elechouse.com/elechouse/index.php?main_page=product_info&cPath=90_93&products_id=2205)

leninhat, 2018-11-04:
The Elechouse pn532 v3 
I can read it using sudo in the command line before nfc-list, but now i can´t read the cards with the iso 14443 2-b or iso 14443b.
i think is another subject, that i´m going to open.

sanbrother, 2019-12-21:
I googled a lot, and finally found that it was due to the power supply. Try another one.