nfc-tools/libnfc #508: Problem during installation

kafffee, 2018-10-10:
Hi!
I am trying to install libnfc on my Raspberry Pi with Raspbian / Debian for use with I2C with the following:

```
git clone https://github.com/nfc-tools/libnfc.git
cd libnfc
git checkout libnfc-1.7.1
git clean -d -f -x
#rm ../libnfc*.deb
git remote|grep -q anonscm||git remote add anonscm git://anonscm.debian.org/collab-maint/libnfc.git
git fetch anonscm
git checkout remotes/anonscm/master debian
git reset
dpkg-buildpackage -uc -us -b
```

When I type
`git fetch anonscm`
I get the following output though:
```
fatal: unable to connect to anonscm.debian.org:
anonscm.debian.org[0: 194.177.211.202]: errno=Verbindungsaufbau abgelehnt
anonscm.debian.org[1: 2001:648:2ffc:deb::211:202]: errno=Das Netzwerk ist nicht erreichbar
```
(which is German and means "connection denied" ("Verbindungsaufbau abgelehnt") respectively "network not available" ("Das Netzwerk ist nicht erreichbar").
 
Also, when I type
`git clean -d -f -x`
and
`#rm ../libnfc*.deb`
and
`git remote|grep -q anonscm||git remote add anonscm git://anonscm.debian.org/collab-maint/libnfc.git`
there is no output at all.

any other ways to install libnfc are welcome, too.

neomilium, 2018-10-10:
To install under Raspian, simply:
```
sudo apt update
sudo apt install libnfc-bin
```

closed 2018-10-10 by neomilium

kafffee, 2018-10-14:
Thanks a lot I got it working.

Whats the easiest way to get a tags UID though?