nfc-tools/libnfc #511: Mac Build

MaxHayman, 2018-10-31:
Trying to build on my new Mac gives

[  1%] Building C object libnfc/CMakeFiles/nfc.dir/drivers/pn532_i2c.c.o
In file included from /Users/x/Code/libnfc/libnfc/drivers/pn532_i2c.c:52:
/Users/x/Code/libnfc/libnfc/buses/i2c.h:43:12: fatal error: 'linux/i2c-dev.h' file
      not found
#  include <linux/i2c-dev.h>
           ^~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [libnfc/CMakeFiles/nfc.dir/drivers/pn532_i2c.c.o] Error 1
make[1]: *** [libnfc/CMakeFiles/nfc.dir/all] Error 2
make: *** [all] Error 2

Although I have run this before on my previous Mac (possibly an older version)

l7777777b, 2018-11-01:
have you tried 'make clean' first?

MaxHayman, 2018-11-01:
Yes. I also tried with the master and the latest release. Same issue.

LudovicRousseau, 2019-04-14:
Do not build with I2C support on macOS.
i2c is for GNU/Linux only, as you discovered.

closed 2019-04-14 by LudovicRousseau