nfc-tools/libnfc #515: nfc_open failed to open device after running nfc-dep-initiator example

Ahmed-Elias, 2018-11-06:
Hello,

I am using PN533 device to run the nfc-dep-initiator sample but once it finishes I am unable to open the device again until replugging the device.

I am using libnfc version 1.7.1

Best regards,
Ahmed

TanukiSharp, 2019-06-01:
Hi @Ahmed-Elias.

Not sure if you are still interested, but I got a similar problem, and solved it by commenting the following line: https://github.com/nfc-tools/libnfc/blob/master/libnfc/chips/pn53x.c#L131

Actually with a Sony reader (Felica) for some reason setting a parameter here fails, but it seems to not cause any problem if the error is ignored. I simply log it but do not return.

I will open/update an issue for that.

Ahmed-Elias, 2019-06-03:
Hello Sebastien,

Thanks for your reply. I will try your solution and feedback you although I
have found another solution that is working fine be me now. after debugging
I found the usb_set_configuration method in pnx53x_usb.c file always fails
after the first connection, I tried first to search if there is a resource
that is not closed correctly but I didn't found so I just now comment the
error handling and it is working fine.

I will try to update if  I find a better solution I working deeply on the
library these days.

Best Regards,
Ahmed