nfc-tools/libnfc #519: nfc-mfclassic can't read or write

mateipopp, 2018-11-12:
Hello guys,
So recently i reinstalled my linux and I'm trying to write/read Mifare classic 1k cards with libnfc.
I have a PN532 Uart and before i reinstalled I could write/reads cards.
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): ab  53  33  f8  
      SAK (SEL_RES): 08  
Guessing size: seems to be a 1024-byte card
Reading out 64 blocks |nfc_initiator_transceive_bytes: Mifare Authentication Failed
nfc_initiator_transceive_bytes: Mifare Authentication Failed
nfc_initiator_transceive_bytes: Mifare Authentication Failed
nfc_initiator_transceive_bytes: Mifare Authentication Failed
nfc_initiator_transceive_bytes: Mifare Authentication Failed
nfc_initiator_transceive_bytes: Mifare Authentication Failed
nfc_initiator_transceive_bytes: Mifare Authentication Failed
nfc_initiator_transceive_bytes: Mifare Authentication Failed
nfc_initiator_transceive_bytes: Mifare Authentication Failed
!
Error: authentication failed for block 0x3f
 This is my error.

kswt, 2018-11-30:
The same problem on Arch linux.

Tyrben, 2019-03-24:
@mateipopp How did you that?

closed 2021-01-04 by doegox