nfc-tools/libnfc #522: error occurs 0xc000007b

oleggy-gvz, 2018-12-02:
Hello. I'm new to using сmake and opensource. I have two questions:
1. In the description of README-Windows.md, it is written "This will be a shared library for Windows. (Nfc.dll)". But it creates file libnfc.dll, not nfc.dll. Is this error in description?
2. When I try to run any exe utilities from \libnfc-build\utils 
system requires "libusb0.dll" library. But when I specify library from \libusb-win32-bin-1.2.6.0\bin\ia64\libusb0.dll launch error occurs 0xc000007b.

daixtrose, 2018-12-02:
Can you please add a complete list of the steps you perform? What command line commands are you using? Which shell are you in? Git Bash, Cygwin? Which compiler (Cygwin, MinGw, Visual Studio) and which version, etc.?

oleggy-gvz, 2018-12-02:
OK. I work for Windows 7 64 bit (Intel CPU). I did on the manual.
1. installed TDM-GCC MinGW Compiler - tdm64-gcc-4.7.1-3, screenshot - https://yadi.sk/i/UZUjXdyLt6LhLg
2. added the PATH directories "c:\MinGW64\bin; c:\MinGW64\x86_64-w64-mingw32\lib32; c:\MinGW64\x86_64-w64-mingw32\include"
3. installed CMake - cmake-3.13.1-win64-x64 and configured CMake, screenshot - https://yadi.sk/i/wUMmJSix6fwsag
4. run mingw32-make to build, screenshot - https://yadi.sk/i/Y7zkajDOBUPd2g
5. compiled libnfc.dll here ..\libnfc-build\libnfc\libnfc.dll
6. downloaded libusb - libusb-win32-bin-1.2.6.0.zip and copied libusb0.dll file from zip - ..\bin\ia64\libusb0.dll
7. I copy files libnfc.dll and libusb0.dll to directory ..\libnfc-build\utils\ and try to run any utility, for example nfc-list.exe

daixtrose, 2018-12-02:
Please use CTRL+v to paste the screenshots directly. Hint: You can edit your answer.

oleggy-gvz, 2018-12-03:
corrected post

chomudda, 2019-06-21:
@oleggy-gvz Did you end up getting it to work? I'm having the exact same issue. As for "LIBNFC_SYSCONFDIR" there is no "config" directory in libnfc. Should it just point to an empty "config" directory?