nfc-tools/libnfc #523: Add support for RC522

goddarkhun, 2018-12-19:
Hey, 
I think it would be really useful for the program to have RC522 support, I have seen multiple treads on forums regarding this question and none of them were solved. I can't do it myself because I don't have the technical knowledge to do it but I am sure someone can.

ildar, 2018-12-19:
this isn't NFC, it is RFID
Though look at https://github.com/nfc-tools/libnfc/pull/300 and
https://github.com/nfc-tools/libnfc/issues/485

goddarkhun, 2018-12-20:
Sorry for the question but could you explain why is that a problem?

ildar, 2018-12-20:
Someone told me that RFIDs aren't welcome here (((
Though I personally think that a proper patch wouldn't hurt the project in
the current state.

goddarkhun, 2018-12-20:
@ildar I don't really understand but OK. I still have to get it to work tho, do you thing it can be done? Or I should buy a pn532?

ildar, 2018-12-21:
I'm not an authority here.
I also think RC522 should work through libnfc but it requires some patching
or development.
If your goals are in RFID scope then it can be done

dfr0, 2019-12-04:
Any update about this?

neomilium, 2021-01-04:
RC522 is a RFID device, not NFC one. Adding it to libnfc will require to implement a capabilities detection…

closed 2021-01-04 by neomilium