nfc-tools/libnfc #524: Buzzer and led usage

MaitreDede, 2018-12-20:
Hi,
I have a ACR122 reader, driver is acr233_usb. I would like to drive its led and buzzer from my application.
Is there any entrypoint in libnfc I can use to achieve this ? Or any other idea ?

Thanks.

ravkr, 2019-06-30:
I've just found that this library had (!) support for led, using the function `acr122_led_red(nfc_device_t *pnd, bool bOn)`
but @smortex commented it out in march 2011 here: https://github.com/nfc-tools/libnfc/commit/88bff13b8571dda322148776c64d2844ca5fb031

You can try to uncomment it, and recompile the library

nobur, 2020-10-24:
Hi,
I know that this is an old post  but for those who are still looking for a workaround to this problem, this is how I did it.
Be careful that this modification will not be supported by the maintainers of this lib and is only valid for ACR122u model.

The best way I found was to expose the send_apdu function. Once done, you'll be only limited by your understanding of the manufacturer documentation (ask google for "ACR122U application programming interface" if you don't have it) .
- edit the libnfc/nfc.c file and add the following function:
```
int
nfc_send_apdu(nfc_device *pnd, const uint8_t ins, const uint8_t p1, const uint8_t p2, const uint8_t *const data,
         size_t data_len, const uint8_t le, uint8_t *out, const size_t out_size)
{
  HAL(send_apdu, pnd,ins,p1,p2,data,data_len,le,out,out_size);
}
```
- edit nfc/nfc.h file and add the function declaration:
```
NFC_EXPORT int nfc_send_apdu(nfc_device *pnd, const uint8_t ins, const uint8_t p1, const uint8_t p2, const uint8_t *const data, size_t data_len, const uint8_t le, uint8_t *out, const size_t out_size);
```
- edit libnfc/nfc-internal.h and insert just before abort_command declaration:
```
int (*send_apdu)(struct nfc_device *pnd, const uint8_t ins, const uint8_t p1, const uint8_t p2, const uint8_t *const data, size_t data_len, const uint8_t le, uint8_t *out, const size_t out_size);
```
- the last step is the more tedeous: edit every driver C source file  (libnfc/drivers/****.c) and add the following line just above abort_command alias ( few lines before the end of the file):
```
.send_apdu      = NULL,
```
except for the "acr122u_usb.c" file where the line should be :
```
  .send_apdu = acr122_usb_send_apdu,
```

remove the lib from your system, recompile and reinstall it as usual. you should now be able to call the nfc_send_apdu function and manage to control buzzer and led state.
Personally, the function I use is the following one:
```
void LNFCReaderBip(nfc_device *device, int8_t type)
{
    if (device == NULL)
        return;
    uint8_t buffRx[255] = {0};
    uint8_t cmd[] = {0X01, 0x00, 0x01, BUZZSTATE}; //default values as a safeguard /  last byte = 1 to enable buzzer

    switch (type)
    {
    case NFCREADER_BIP_OK:
        cmd[1] = 0x00;
        cmd[2] = 0x01;
        break;
    case NFCREADER_BIP_LONG:
        cmd[0] = 0x05;
        break;
    case NFCREADER_BIP_THREE_SHORT:
        cmd[1] = 0x01;
        cmd[2] = 0x03;
        break;
    case NFCREADER_BIP_INITREADER:
        cmd[1] = 0X01;
        cmd[2] = 0x02;
        break;
    }

    int prevCancelState = 0;
    pthread_setcancelstate(PTHREAD_CANCEL_DISABLE, &prevCancelState); //prevent thread interruption during processing
    nfc_send_apdu(device, 0x00, 0x40, 0x50, (uint8_t *)cmd, sizeof(cmd), 0, buffRx, sizeof(buffRx));
    pthread_setcancelstate(prevCancelState, NULL);
}
```

Be careful that if the sum of cmd[1] and cmd[2] exceeds 4 you may face an issue with some revision of the reader. In version 1.8.0 of the lib, the timeout delay for the reader to answer the command is set to 1 second. if the Led/buzzer cycle is longer than that, the answer will be received after the timeout and will be interpreted as the answer to the next command. Then you'll probably get a message similar to the one hereunder, followed by a reader reset.
```
error	libnfc.driver.acr122_usb	Unexpected Status Word (SW1: 90 SW2: 02)
error	libnfc.driver.acr122_usb	Command Code verification failed
```
To solve this, you can simply choose lower values for durations cmd[1] & cmd[2] (prefered solution) or increase the timeout if the acr122_usb_bulk_read call in acr122_usb_send_apdu function of the libnfc/drivers/arc122_usb.c file. as followed : 
```
change
if ((res = acr122_usb_bulk_read(DRIVER_DATA(pnd), out, out_size, 1000)) < 0)
to
if ((res = acr122_usb_bulk_read(DRIVER_DATA(pnd), out, out_size, 1500)) < 0)
```
**changing timeout to a too high value can have a significant incidence on the lib general behavior.**

Hope this could help someone

closed 2021-01-04 by smortex