nfc-tools/libnfc #526: Can't build on macOS 10.14.4

jukben, 2019-01-19:
I got this. However, `autoreconf -is` and `./configuration` was without any hassle.
Anyone was facing the same issue? I'd love to be pointed out. 🙏 
```
libnfc master ✓ make
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in libnfc
Making all in chips
  CC       libnfcchips_la-pn53x.lo
  CCLD     libnfcchips.la
ar: `u' modifier ignored since `D' is the default (see `U')
Making all in buses
  CC       libnfcbuses_la-empty.lo
  CC       libnfcbuses_la-uart.lo
  CCLD     libnfcbuses.la
ar: `u' modifier ignored since `D' is the default (see `U')
Making all in drivers
  CC       libnfcdrivers_la-pn532_uart.lo
  CCLD     libnfcdrivers.la
ar: `u' modifier ignored since `D' is the default (see `U')
Making all in .
  CC       libnfc_la-conf.lo
  CC       libnfc_la-iso14443-subr.lo
  CC       libnfc_la-mirror-subr.lo
  CC       libnfc_la-nfc.lo
  CC       libnfc_la-nfc-device.lo
  CC       libnfc_la-nfc-emulation.lo
  CC       libnfc_la-nfc-internal.lo
  CC       libnfc_la-target-subr.lo
  CC       libnfc_la-log.lo
  CC       libnfc_la-log-internal.lo
  CCLD     libnfc.la
/usr/local/bin/nm: libnfcbuses_la-empty.o: no symbols
ar: `u' modifier ignored since `D' is the default (see `U')
Making all in utils
  CC       nfc-barcode.o
  CC       nfc-utils.lo
  CCLD     libnfcutils.la
ar: `u' modifier ignored since `D' is the default (see `U')
  CCLD     nfc-barcode
ld: warning: ignoring file ./.libs/libnfcutils.a, file was built for archive which is not the architecture being linked (x86_64): ./.libs/libnfcutils.a
  CC       nfc-emulate-forum-tag4.o
  CCLD     nfc-emulate-forum-tag4
ld: warning: ignoring file ./.libs/libnfcutils.a, file was built for archive which is not the architecture being linked (x86_64): ./.libs/libnfcutils.a
Undefined symbols for architecture x86_64:
  "_print_hex", referenced from:
      _nfcforum_tag4_io in nfc-emulate-forum-tag4.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [nfc-emulate-forum-tag4] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2```

LudovicRousseau, 2019-04-14:
No problem here on macOS 10.4.4.
Maybe you can use "make clean" to recompile everything or "make_release.sh" to regenerate the autotools files.

closed 2019-04-14 by LudovicRousseau

jukben, 2019-04-14:
Could you please summarize all the necessary steps to successfully build it?

- autoreconf -is
- `./configure`
- make clean
- make

and I still get this error above

LudovicRousseau, 2019-04-14:
Does all the steps you mention finish with *no* error?

jukben, 2019-04-15:
Yes. Just the “make” ends up with the error above.

LudovicRousseau, 2019-04-15:
Remove the `.libs` directory and try again.

TachikakaMin, 2020-08-19:
same problem in macOS 10.14.6

LudovicRousseau, 2020-08-21:
Try again using `libnfc-1.8.0.tar.bz2` from https://github.com/nfc-tools/libnfc/releases/tag/libnfc-1.8.0