nfc-tools/libnfc #530: `PKG_CHECK_MODULES(libnfc, libnfc >= $LIBNFC_REQUIRED_VERSION, , AC_MSG_ERROR([libnfc >= $LIBNFC_REQUIRED_VERSION is mandatory.]))'

Elkhazrajy, 2019-02-17:
I Can't compile mfcuk because of this error

root@kali:~/mfcuk# ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking whether make supports nested variables... (cached) yes
./configure: line 3960: syntax error near unexpected token `libnfc,'
./configure: line 3960: `PKG_CHECK_MODULES(libnfc, libnfc >= $LIBNFC_REQUIRED_VERSION, , AC_MSG_ERROR([libnfc >= $LIBNFC_REQUIRED_VERSION is mandatory.]))'

please give me a solution

majidf, 2019-03-11:
run ''autoreconf'' -is before ./configure.

Elkhazrajy, 2019-03-11:
@majidf I did but nothing has changed, it gives me same error

ildar, 2019-03-12:
It seems that autoreconf doesn't see m4 files from libnfc. You should have
libnfc devel files installed correctly or use `-I`

LudovicRousseau, 2019-04-14:
install the `pkg-config` package to get the `PKG_CHECK_MODULES` macro definition.

closed 2019-04-14 by LudovicRousseau

Elkhazrajy, 2019-04-14:
@LudovicRousseau this package (pkg-config) already installed

LudovicRousseau, 2019-04-14:
Make sure `autoreconf -is` works with _no_ error to generate a correct configure script.