nfc-tools/libnfc #532: acr122u RF Transmission Error and NTAG215 are detected as a Mifare Ultralight when using Raspberry PI

4ndv, 2019-02-22:
When I'm connecting ACR122U to the Raspberry PI 3 (raspbian, latest libnfc, latest libfreefare), it cannot detect NTAG215 (ntag-detect returns Mifare Ultralight, but when I'm using my Linux laptop everything is fine.

Full log:
```
➜  ~ sudo LIBNFC_LOG_LEVEL=3 ntag-detect
info	libnfc.config	Unable to open file: /usr/local/etc/nfc/libnfc.conf
debug	libnfc.config	Unable to open directory: /usr/local/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
debug	libnfc.driver.acr122_usb	device found: Bus 001 Device 007 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:001:007"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 3b 00
debug	libnfc.driver.acr122_usb	ACR122 PICC Operating Parameters
debug	libnfc.driver.acr122_usb	TX: 6f 05 00 00 00 00 00 00 00 00 ff 00 51 00 00
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 90 00
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00
debug	libnfc.chip.pn53x	SetParameters
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 12 14
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 13 90 00
debug	libnfc.general	"ACS / ACR122U PICC Interface" (acr122_usb:001:007) has been claimed.
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 11 00 00 00 00 00 00 00 00 ff 00 00 00 0c d4 06 63 02 63 03 63 0d 63 38 63 3d
debug	libnfc.driver.acr122_usb	RX: 80 09 00 00 00 00 00 00 81 00 d5 07 a2 a2 00 00 00 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 01
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 ff ff ff
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 13 00 00 00 00 00 00 00 00 ff 00 00 00 0e d4 06 63 02 63 03 63 05 63 38 63 3c 63 3d
debug	libnfc.driver.acr122_usb	RX: 80 0a 00 00 00 00 00 00 81 00 d5 07 a2 a2 00 00 10 00 90 00
debug	libnfc.chip.pn53x	PN53X_REG_CIU_TxMode (Defines the transmission data rate and framing during transmission)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_RxMode (Defines the transmission data rate and framing during receiving)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_TxAuto (Controls the settings of the antenna driver)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 10 00 00 00 00 00 00 00 00 ff 00 00 00 0b d4 08 63 02 80 63 03 80 63 05 40
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 01
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 00 01 02
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
debug	libnfc.chip.pn53x	InListPassiveTarget
debug	libnfc.chip.pn53x	No timeout
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 4a 01 00
debug	libnfc.driver.acr122_usb	RX: 80 11 00 00 00 00 00 00 81 00 d5 4b 01 01 00 44 00 07 04 c9 10 aa 84 49 80 90 00
debug	libnfc.chip.pn53x	InDeselect
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 44 00
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 45 00 90 00
debug	libnfc.chip.pn53x	InListPassiveTarget
debug	libnfc.chip.pn53x	No timeout
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 4a 01 00
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 4b 00 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 ff ff ff
[... 39 lines cut ...]
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
debug	libnfc.chip.pn53x	InListPassiveTarget
debug	libnfc.chip.pn53x	No timeout
debug	libnfc.driver.acr122_usb	TX: 6f 0e 00 00 00 00 00 00 00 00 ff 00 00 00 09 d4 4a 01 02 00 ff ff 01 00
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 4b 00 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 ff ff ff
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
Tag with UID 04c910aa844980 is a Mifare UltraLight
debug	libnfc.driver.acr122_usb	ACR122 Abort
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00
debug	libnfc.chip.pn53x	InRelease
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 52 00
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 53 00 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
```

Then only difference I can see with the log from laptop is these errors:

```
debug	libnfc.chip.pn53x	Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20))
```

4ndv, 2019-02-25:
Also tried RPi 1 B+ with the same latest raspbian, same result. May it be some kind of problem in the raspbian or it's a reader itself?

neomilium, 2021-01-04:
Did you check the libnfc version on both OSes?

closed 2021-01-04 by neomilium