nfc-tools/libnfc #533: Unable to open directory: /etc/nfc/devices.d

maythayus, 2019-02-22:
libnfc-master$ sudo LIBNFC_LOG_LEVEL=3 nfc-list -v
debug   libnfc.config   Unable to open directory: /etc/nfc/devices.d
debug   libnfc.general  log_level is set to 3
debug   libnfc.general  allow_autoscan is set to true
debug   libnfc.general  allow_intrusive_scan is set to false
debug   libnfc.general  0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug   libnfc.driver.acr122_usb        device found: Bus 001 Device 015 Name ACS ACR122
debug   libnfc.general  1 device(s) found using acr122_usb driver
debug   libnfc.general  0 device(s) found using pn53x_usb driver
debug   libnfc.driver.acr122_usb        3 element(s) have been decoded from "acr122_usb:001:015"
error   libnfc.driver.acr122_usb        Unable to set alternate setting on USB interface (Connection timed out)
debug   libnfc.general  Unable to open "acr122_usb:001:015".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:001:015

maythayus, 2019-02-22:
Unable to set alternate setting on USB interface

maythayus, 2019-02-22:
plz help me ?

maythayus, 2019-02-22:
libnfc-master$ sudo LIBNFC_LOG_LEVEL=3 nfc-list -v
debug   libnfc.general  log_level is set to 3
debug   libnfc.general  allow_autoscan is set to true
debug   libnfc.general  allow_intrusive_scan is set to false
debug   libnfc.general  0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug   libnfc.driver.acr122_usb        device found: Bus 001 Device 015 Name ACS ACR122
debug   libnfc.general  1 device(s) found using acr122_usb driver
debug   libnfc.general  0 device(s) found using pn53x_usb driver
debug   libnfc.driver.acr122_usb        3 element(s) have been decoded from "acr122_usb:001:015"
error   libnfc.driver.acr122_usb        Unable to set alternate setting on USB interface (Connection timed out)
debug   libnfc.general  Unable to open "acr122_usb:001:015".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:001:015

remytms, 2019-08-09:
@maythayus Your error seams to look similar to #535. Did you use the new ACR122U-A9 nfc reader from Advanced Card Systems ?
I'm facing the same error with this device.

neomilium, 2021-01-04:
ACR122U is bogus.

closed 2021-01-04 by neomilium