nfc-tools/libnfc #534: No NFC device found

ChunJyeBehBeh, 2019-03-10:
Hello,

I have connect my PN532 to raspberry pi 3B+ with I2C wiring. 

I follow all the installation but still cannot use it 

after I run `nfc-list` , the error shown:
```
nfc-list uses libnfc 1.7.1
No NFC device found.
```

Below is the output after I run `LIBNFC_LOG_LEVEL=3 nfc-scan-device -i -v` ,
```
info	libnfc.config	Unable to open file: /etc/nfc/libnfc.conf
debug	libnfc.config	key: [device.allow_autoscan], value: [true]
info	libnfc.config	Unknown key in config line: device.allow_autoscan = true
debug	libnfc.config	key: [device.allow_intrusive_scan], value: [false]
info	libnfc.config	Unknown key in config line: device.allow_intrusive_scan = false
debug	libnfc.config	key: [device.log_level], value: [2]
info	libnfc.config	Unknown key in config line: device.log_level = 2
debug	libnfc.config	key: [device.device.name], value: [PN532 NFC]
info	libnfc.config	Unknown key in config line: device.device.name = PN532 NFC
debug	libnfc.config	key: [device.device.connstring], value: [pn532_uart:/dev/ttyAMA0]
info	libnfc.config	Unknown key in config line: device.device.connstring = pn532_uart:/dev/ttyAMA0
debug	libnfc.config	key: [device.device.name], value: [PN532 board via UART]
info	libnfc.config	Unknown key in config line: device.device.name = PN532 board via UART
debug	libnfc.config	key: [device.allow_intrusive_scan], value: [true]
info	libnfc.config	Unknown key in config line: device.allow_intrusive_scan = true
debug	libnfc.config	key: [device.device.connstring], value: [pn532_i2c:/dev/i2c-1]
info	libnfc.config	Unknown key in config line: device.device.connstring = pn532_i2c:/dev/i2c-1
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
nfc-scan-device uses libnfc 1.7.1
debug	libnfc.driver.arygon	Trying to find ARYGON device on serial port: /dev/ttyAMA0 at 9600 bauds.
debug	libnfc.driver.arygon	Trying to find ARYGON device on serial port: /dev/ttyS0 at 9600 bauds.
debug	libnfc.general	0 device(s) found using arygon driver
debug	libnfc.driver.pn532_i2c	Trying to find PN532 device on I2C bus /dev/i2c-1.
debug	libnfc.chip.pn53x	Diagnose
debug	libnfc.chip.pn53x	Timeout value: 500
debug	libnfc.chip.pn53x	SAMConfiguration
debug	libnfc.chip.pn53x	Timeout value: 1000
debug	libnfc.bus.i2c	TX: 00 00 ff 03 fd d4 14 01 17 00 
error	libnfc.bus.i2c	Error: wrote only -1 bytes (10 expected).
error	libnfc.driver.pn532_i2c	Unable to transmit data. (TX)
debug	libnfc.general	0 device(s) found using pn532_i2c driver
debug	libnfc.general	0 device(s) found using pn532_spi driver
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyAMA0 at 115200 bauds.
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyS0 at 115200 bauds.
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.driver.acr122s	Trying to find ACR122S device on serial port: /dev/ttyAMA0 at 9600 bauds.
debug	libnfc.driver.acr122s	Trying to find ACR122S device on serial port: /dev/ttyS0 at 9600 bauds.
debug	libnfc.general	0 device(s) found using ACR122S driver
debug	libnfc.general	0 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
No NFC device found.
```

and the content inside my pn532_uart_on_rpi.conf  file,
```
device.name = "PN532 board via UART"
allow_intrusive_scan = true
device.connstring = pn532_i2c:/dev/i2c-1
```

Can anyone help me to solve me? Thanks.

Clinan, 2019-05-11:
@ChunJyeBehBeh  
hello Chun, Did you resolve this issues?

romain145, 2020-10-15:
Fix on RPI 3:

1. Make sure the PN532 is present:
```
pi@raspberrypi:~/libnfc$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- 24 -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
2. Configure libnfc with the I2C driver and install:
```
./configure --with-drivers=pn532_i2c
make
sudo make install
```
3. Fix Missing shared libraries after install #615:
```
sudo sh -c "echo /usr/local/lib > /etc/ld.so.conf.d/usr-local-lib.conf"
sudo ldconfig
```

4. Create /usr/local/etc/nfc/libnfc.conf with this content:
```
device.name = "PN532 board via I2C"
allow_intrusive_scan = true
device.connstring = pn532_i2c:/dev/i2c-1
```

5. And voilà:
```
pi@raspberrypi:~ $ LIBNFC_LOG_LEVEL=3 nfc-scan-device -i -v
debug   libnfc.config   key: [device.name], value: [PN532 board via UART]
debug   libnfc.config   key: [allow_intrusive_scan], value: [true]
debug   libnfc.config   key: [device.connstring], value: [pn532_i2c:/dev/i2c-1]
debug   libnfc.config   Unable to open directory: /usr/local/etc/nfc/devices.d
debug   libnfc.general  log_level is set to 3
debug   libnfc.general  allow_autoscan is set to true
debug   libnfc.general  allow_intrusive_scan is set to true
debug   libnfc.general  1 device(s) defined by user
debug   libnfc.general    #0 name: "PN532 board via UART", connstring: "pn532_i2c:/dev/i2c-1"
nfc-scan-device uses libnfc libnfc-1.8.0-43-gc3f739d
debug   libnfc.driver.pn532_i2c Trying to find PN532 device on I2C bus /dev/i2c-1.
debug   libnfc.chip.pn53x       Diagnose
debug   libnfc.chip.pn53x       Timeout value: 500
debug   libnfc.chip.pn53x       SAMConfiguration
debug   libnfc.chip.pn53x       Timeout value: 1000
debug   libnfc.bus.i2c  TX: 00 00 ff 03 fd d4 14 01 17 00
debug   libnfc.bus.i2c  wrote 10 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.bus.i2c  TX: 00 00 ff 09 f7 d4 00 00 6c 69 62 6e 66 63 be 00
debug   libnfc.bus.i2c  wrote 16 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.general  1 device(s) found using pn532_i2c driver
2 NFC device(s) found:
debug   libnfc.chip.pn53x       Diagnose
debug   libnfc.chip.pn53x       Timeout value: 500
debug   libnfc.chip.pn53x       SAMConfiguration
debug   libnfc.chip.pn53x       Timeout value: 1000
debug   libnfc.bus.i2c  TX: 00 00 ff 03 fd d4 14 01 17 00
debug   libnfc.bus.i2c  wrote 10 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.bus.i2c  TX: 00 00 ff 09 f7 d4 00 00 6c 69 62 6e 66 63 be 00
debug   libnfc.bus.i2c  wrote 16 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
[... 97 lines cut ...]
Mifare tag present:
```
pi@raspberrypi:~ $ nfc-list
NFC device: PN532 board via UART opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): e0  11  d3  1b
      SAK (SEL_RES): 08

error   libnfc.driver.pn532_i2c Application level error detected  (127)
error   libnfc.driver.pn532_i2c Application level error detected  (127)
NFC device: PN532 board via UART opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): e0  11  d3  1b
      SAK (SEL_RES): 08

error   libnfc.driver.pn532_i2c Application level error detected  (127)
error   libnfc.driver.pn532_i2c Application level error detected  (127)
```

1. I don't know why it mentions UART while it's using I2C
2. The errors are mentioned here https://github.com/nfc-tools/libnfc/issues/493, no response but it seems to work (?!)

neomilium, 2021-01-04:
@romain145  Thanks for explanations :+1:

closed 2021-01-04 by neomilium

Donnadonnam, 2023-03-11:
Haha bitches I'm the creator of NFC fuck you all!!!