nfc-tools/libnfc #535: Unable to set alternate setting on USB interface (Connection timed out)

alexandredubois, 2019-03-28:
Hello,

I get the following message using my ACR_122U

```
$ sudo nfc-scan-device -v
nfc-scan-device uses libnfc 1.7.1
1 NFC device(s) found:
error   libnfc.driver.acr122_usb        Unable to set alternate setting on USB interface (Connection timed out)
nfc_open failed for acr122_usb:003:005
```
Is someone familiar with this problem ? Any idea of what i can look for to troubleshoot this ?

gdw96, 2019-03-30:
I have the same problem.

I must have already started with blacklist pn533_usb.
Then before each use of my RFID reader I have to do "service pcscd restart" and then the reader starts working (LED lights up, beeps, etc.)

But I have exactly the same problem and no solution.

mondonc, 2019-04-04:
I have the same problem too.

With debian stretch, libnfc and pkg-config installed for compilation from source : 
dpkg -l | grep nfc
ii  libnfc-bin                           1.7.1-4+b1
ii  libnfc-dev:amd64                     1.7.1-4+b1
ii  libnfc5:amd64                        1.7.1-4+b1 

cat /etc/modprobe.d/blacklist-libnfc.conf 
blacklist nfc
blacklist pn533

not enough after rebbot, I have to manually run 
modprobe -r pn533_usb
modprobe -r pn533
modprobe -r nfc

But it still does not work.

Not any pcscd service on my system.

I have not ideas, anybody ?

dragonlost, 2019-04-05:
i have exact same problem !

```
$ nfc-list
nfc-list uses libnfc 1.7.1
error	libnfc.driver.acr122_usb	Unable to set alternate setting on USB interface (Connection timed out)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:001:003
sdurand@portable_seb:~$ LIBNFC_LOG_LEVEL=3 nfc-list -v
debug	libnfc.config	Unable to open directory: /etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug	libnfc.driver.acr122_usb	device found: Bus 001 Device 003 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:001:003"
error	libnfc.driver.acr122_usb	Unable to set alternate setting on USB interface (Connection timed out)
debug	libnfc.general	Unable to open "acr122_usb:001:003".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:001:003

```

i have install rules --> no effect
i have install pcscd --> no effect
i have add pn533_usb in blacklist  --> no effect

my config : Ubuntu Mate 18.04.2 LTS Amd64 and 4.15.0-45-generic

dragonlost, 2019-04-05:
But if I run the service pcscd so, the player work and biip on my nfc tags.

rosslev, 2019-04-16:
same problem

filsif, 2019-04-25:
I have the same problem.

Ubuntu 18.04 , and I tried debian 9.8.0 liveCD.

There is a kernel message after calling nfc-scan-device -v :

usbfs : process 3206 ( pcscd ) did not claim interface 0 before use

Maybe a wrong version of library ? I can use any linux distrib, which version is easy to use acr122 ?

Najihel, 2019-05-09:
Hello,

I have the same problem too

sublimz, 2019-05-12:
Hello,
same problem for me on fresh Kali live

blafois, 2019-05-13:
Commenting line 433 of acr122_usb.c, replacing it by "res = 0;" seems to fix the issue on my machine.

sublimz, 2019-05-13:
Yes !,
Commenting line 433 of acr122_usb.c, replacing it by "res = 0;"
resolve my problem too ;
thanks @blafois

Greg270, 2019-05-14:
Hello,

I'm a novice.
How you can edit the line 433 in acr122_usb.c ? what command do you use ? thanks !

Greg270, 2019-05-20:
Up Please, what is the steps to edit the line ?

blafois, 2019-05-20:
> Up Please, what is the steps to edit the line ?

You need to clone the repo, edit the file aforementionned, recompile the lib.

sublimz, 2019-05-20:
Edit : for me, the fix works in read but i can't write, I try it
with nfc-mfclassic.
I try another new acr122u device and i can read and write without fix !

juanjo666, 2019-05-21:
@sublimz entonces que camino hay que tomar? por mi parte he probado en distintas distribuciones, hasta con la version 1.7.0 de libnfc y mismo error, estoy sospechando que sea el propio lector, a lo cual no le encuentro explicacion, alguna sugerencia o alternativa para poder darle paso de una vez por todas a mfcuk?

Kerberos73, 2019-05-25:
Hi there. Also noob. Have now cloned the git repository "libnfc.git" - and I found the acr122_usb.c file and did edit it. But cant find out how to compile the files in Kali... Any suggesions?

Kerberos73, 2019-05-25:
Hi! Have tried to change the value in the file, now I still get the error, but it has changed:

error libnfc.driver.acr122_usb Unable to set alternate setting on USB interface (Success)
mfcuk: ERROR: connecting to NFC reader

Any clues?

martinayotte, 2019-06-05:
I'm facing the same issue ...
I've try commenting line 433 where there is "usb_close(data.pudh);" and replacing it with "res =0;" but also commenting the "goto free_mem;" since it doesn't make sense to leave that there, but no success :
It is now showing the following error : 
```
# /usr/local/bin/nfc-scan-device -v
/usr/local/bin/nfc-scan-device uses libnfc libnfc-1.7.1-215-gc42e250
1 NFC device(s) found:
error	libnfc.driver.acr122_usb	Unable to set alternate setting on USB interface (Connection timed out)
error	libnfc.driver.acr122_usb	Unable to write to USB (No such file or directory)
nfc_open failed for acr122_usb:008:004
```

StephanCandA, 2019-06-12:
I also have the same issue. I tried it on a Raspberry (Raspian) and on a PC (Debian) both with a 4.19 + kernel.

my log from the pc:

sudo LIBNFC_LOG_LEVEL=3 nfc-list
debug	libnfc.config	key: [allow_autoscan], value: [true]
debug	libnfc.config	Unable to open directory: /etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug	libnfc.driver.acr122_usb	device found: Bus 006 Device 002 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:006:002"
error	libnfc.driver.acr122_usb	Unable to set alternate setting on USB interface (Connection timed out)
debug	libnfc.general	Unable to open "acr122_usb:006:002".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:006:002

during this log the pcscd service is stopped but I get the same result with the service started.

Slon4564556, 2019-07-20:
Hi,
The previous answers did not work with my acr122u, in kali live or ubuntu 18.04.2. New ways to solve this problem?

RCTORONTO, 2019-07-23:
I also have this issue, using a acr122u-a9, Ubuntu 17.04, tried the current libnfc and the previous release, I can get the pcsc driver to work, but following the instructions to recompile with acr122_usb, and blacklisting the modules, and stopping pcscd results in the connection time out error - is this a problem with just the A9 variant of the device, if so I will order an older version.  Has anyone found a solution?

poolochon, 2019-08-16:
Same problem with ACR122U-A9.

lelkluk, 2019-08-20:
The same here with Kali, no matter the version, with pn533_usb and pn533 blacklisted :(

arall, 2019-08-23:
Same problem here

harriperhana, 2019-08-29:
I'm having the same issue, is there going to be fix for this?

Clemens-Dautermann, 2019-08-30:
I got the same issue

alexandredubois, 2019-09-10:
A small update 6 months later :
- The fix provided by @jpwidera seems to solve the problem... if you are able to compile the sourcecode yourself.
- After discovering this issue, i choosed to buy my new ACR122U-A9 units directly from ACS and i did not notice this faulty behaviour anymore !

lelkluk, 2019-09-10:
Here is the fastest workaround I managed to do with a live USB key
https://debian-facile.org/viewtopic.php?pid=310787#p310787  msg#37

Clemens-Dautermann, 2019-10-02:
> Here is the fastest workaround I managed to do with a live USB key
> https://debian-facile.org/viewtopic.php?pid=310787#p310787 msg#37

This workaround does actually work for me. Thanks a lot!

alexandredubois, 2019-10-03:
For french readers, i wrote a blog post about the full "Unable to set alternate setting on USB interface" story with a solution to make this ACR122U to work on Kali latest version.

For non-french readers, i guess you will be able to pick the right commands :)

https://www.latelierdugeek.fr/2019/09/30/acr122u-resoudre-lerreur-unable-to-set-alternate-setting-on-usb-interface/

sublimz, 2019-10-03:
Thanks Alexandre, i read your post. Are you trying to write NFC device with
the fix?  I can read but not write.

alexandredubois, 2019-10-03:
I wrote and read various tags with the fix. Can you share the error you
encountered?

qutorial, 2019-11-10:
Hi, if you haven't bother to read the explanations in French.
This fix works: https://github.com/jpwidera/libnfc/
https://github.com/jpwidera/libnfc/commit/a77a2a84971f4c588a6377b30975bbe3b15d5ab4
For me it was a clone from there and a recompile:
`
./configure --with-drivers=acr122s,acr122_pcsc,acr122_usb  --sysconfdir=/etc --prefix=/usr
make
sudo make install
`

5trohm, 2020-01-04:
building the drivers new without the alternative USB setting case wouldn't work for me...
got the same wrong behavior as before. seems like it timed out and then got a connection right after setting the error message... :(

`root@kali:~# nfc-list
nfc-list uses libnfc 1.7.1
error   libnfc.driver.acr122_usb        Unable to set alternate setting on USB interface (Connection timed out)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:002:003
NFC device: ACS ACR122U 00 00 / ACR122U215 opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 8e  53  c9  ab  
      SAK (SEL_RES): 08  

`

thekix, 2020-01-04:
> building the drivers new without the alternative USB setting case wouldn't work for me...
> got the same wrong behavior as before. seems like it timed out and then got a connection right after setting [...]

Hi,

are you using the compiled libraries? Perhaps old libraries are in the system. (type and ldd may help)

Regards,
kix

ganzgustav22, 2020-03-27:
I just experienced the issue on a Raspberry Pi with Raspbian (based on Debian 10 Buster) and libnfc 1.7.1.

Tomorrow, this issue will be one year old. Is the fix by jpwidera here: https://github.com/jpwidera/libnfc/commit/a77a2a84971f4c588a6377b30975bbe3b15d5ab4 still the way to go?

closed 2020-04-05 by AdamLaurie

SpyroTEQ, 2020-06-19:
Hi,
Fix seems already shipped with release 1.8.0 (thanks contributors !)

Since my apt repositories does not have the up-to-date version of libnfc (only 1.7.1), then I took the release 1.8.0 from github, extracted it, and `./configure ; make ; make install` (no source code change needed)

Now `nfc-list` uses the 1.8.0 version, and the "Unable to set..." issue disappeared (@alexandredubois c'est un RR171-430687 que j'ai, donc ça concerne aussi certains RR171-4* [au moins] )

AlexeyAV87, 2020-10-08:
Hi frends, i resolve this question !!!!

0) put packages, the seller sent from aliexpress.com package list (I buy ACR122U-A9 from aliexpress.com )
apt-get install pcsc-tools
apt-get install pcscd
apt-get install libacsccid1
apt-get install --reinstall libpcsclite1

1) Unpack the libnfc-1.7.1.tar archive to the root of the disk /

2) Install the libusb-dev package from the distribution repositories

3) We edit the file /libnfc-1.7.1/libnfc/drivers/acr122_usb.c, find and comment on this line, there should be line 430:

/*
     res = usb_set_altinterface(data.pudh, 0);
      if (res < 0) {
        log_put(LOG_GROUP, LOG_CATEGORY, NFC_LOG_PRIORITY_ERROR, "Unable to set alternate setting on USB interface (%s)", _usb_strerror(res));
        usb_close(data.pudh);
        // we failed to use the specified device
        goto free_mem;
      }
*/

4) Excecute
./configure --prefix=/usr --sysconfdir=/etc --with-drivers=acr122_pcsc,acr122_usb,acr122s,arygon

5) Making sure the configuration is correct

Selected drivers:
   acr122_pcsc...... yes
   acr122_usb....... yes
   acr122s.......... yes
   arygon........... yes
   pn53x_usb........ no
   pn532_uart....... no
   pn532_spi.......  no
   pn532_i2c........ no

6) If everything is fine, do
make

7) If everything is fine, do
make install

8) We can specify an explicit path for the location of libnfc.so
export LD_LIBRARY_PATH=/usr/lib

9) Editing file:

/etc/modprobe.d/blacklist-libnfc.conf

should be the following, add / change 

#blacklist nfc
blacklist pn533
blacklist pn533_usb

10) Excecute

modprobe nfc

11) pcscd - should be off
killall -9 pcscd

12) if there is something in the output of lsmod | grep pn
bang, modprobe pn533_usb -r, should be empty lsmod | grep pn

marcin-chwedczuk, 2020-11-28:
It's actually a huge pain to compile from sources. 
I needed to install `libusb-dev` package (you may also need to install `libusb-0.1` depending on your sys config), then I executed `cmake` and `make` and `make install`.

After `make install` the lib was put into `/usr/lib/local` and was not on my LD_LIBRARY_PATH.
I have to set it up at hoc:
```
$ sudo -i
root@darkness:~# export LD_LIBRARY_PATH=/usr/local/lib/
root@darkness:~# nfc-list
nfc-list uses libnfc 1.8.0
NFC device: ACS / ACR122U PICC Interface opened
root@darkness:~# nfc-poll 
nfc-poll uses libnfc 1.8.0
NFC reader: ACS / ACR122U PICC Interface opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 8b  3f  d6  3d  
      SAK (SEL_RES): 08
```
After that the device worked but the LED and Sound where not generated while reading cards...
Probably I need to spend more time on it...

Kinny092, 2025-08-11:
it is still not working. Tried to comment out the whole function in acr122_usb.c as well as to change the res > 0 operator to res = 0. None works. 

`nfc-list uses libnfc 1.7.1
error	libnfc.driver.acr122_usb	Unable to write to USB (Connection timed out)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:002:007`

`libusb-dev is already the newest version (2:0.1.12-32)
libusb-0.1-4 is already the newest version (2:0.1.12-32)`

PR #561 (2020-04-05): Added check for USB alternate settings
Quick and dirty fix #535 

I'm not sure but I think, that the 072f:2200-device does not support alernative usb modes:
```shell
Bus 001 Device 013: ID 072f:2200 Advanced Card Systems, Ltd ACR122U:

# lsusb -v -s 001:013 | grep bAlter
     bAlternateSetting       0
```
Disabling allows to read and write:
```
./nfc-mfclassic r u a test.mfd
./nfc-mfclassic w u a test.mfd
```

I'm not sure. But does any of these acr122-devices has an alternative usb mode?