nfc-tools/libnfc #536: nfc-poll does not time out/fail when read reader is disconnected

falkorichter, 2019-04-05:
We are running nfc-poll on a raspberry pi compute module product, basically a raspberry pi.

The nfc reader is connected via SPI.

```
pi@zwavetest003:~ $ time nfc-poll
nfc-poll uses libnfc 1.7.1
NFC reader: pn532_spi:/dev/spidev0.0 opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
nfc_initiator_poll_target: Success

real	0m36.281s
user	0m6.930s
sys	0m29.040s
```
no nfc tag connected - `nfc-poll` exits after ~30seconds as expected

```
pi@zwavetest003:~ $ time nfc-poll
nfc-poll uses libnfc 1.7.1
NFC reader: pn532_spi:/dev/spidev0.0 opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
```
I physically disconnect the reader from the SPI now.
```
^Cnfc_initiator_poll_target: Success
error	libnfc.driver.pn532_spi	Unable to wait for SPI data. (RX)

real	4m14.666s
user	0m50.060s
sys	3m21.550s
```
After 4 minutes the `nfc-poll`  finally exits.

Why does it not exit before?

falkorichter, 2019-04-13:
@informaticore ping

neomilium, 2021-01-04:
Run it with debug and dig to identify the latency source. It could be related to low-level access.
Happy hacking ;-)

closed 2021-01-04 by neomilium