nfc-tools/libnfc #540: Support for other chips/devices.

henri62, 2019-05-06:
Question:
Does anyone have an idea if completely other chips can be supported by libnfc?
I'm especially interested in the CR95HF transceiver (UART host interface), I want to make an estimation how much work it is to implement this IC in libnfc.

Is it even possible to do this?
Are there any special features in the pn53x device so it prevents me to do it?

ildar, 2019-05-08:
It's definitely possible. One incomplete example is
https://github.com/nfc-tools/libnfc/pull/300 . There may be better examples
around.

hdeveer, 2019-05-13:
1) Can somebody explain the use of the chip v.s. drivers directory? It look s like the "chips" directory is badly named or should not exist at all?
If I look into all sources it looks like there was made an attempt to separate the chip logic from the transport method (uart/i2c/usb etc), but incompletely done (like the acr122s).

2) The commonality between all drivers in in "struct nfc_driver". Is there an explanation what each function is supposed to do? Now it's  a complete puzzle to find it out. I even have no clue if I don't have a severe hardware mismatch to implement another chip.

ildar, 2019-05-13:
Guess you want to chat about it?
Come to the #libnfc channel at freenode IRC