nfc-tools/libnfc #547: New ACR122U-A9 do not work with libnfc, report different PICC interface.

DmitriR, 2019-06-12:
I have just got several brand-new ACR122U-A9 units; quite unexpectedly, they report a different PICC  interface, and do not work (do not read any cards, and hang / never exit from nfc-list). The "old" units I have work as expected. Are these "new" units actually fakes, or some undocumented internal change by the manufacturer that breaks libnfc (or requires reconfiguring / recompiling?) 

I really need these new units to work, so help is much appreciated. 

The external appearance appear identical except for a thicker plastic on the USB plug on the "new" units; the information labels look identical textually except for the serial numbers (the "old" are lower: RR171-323xxx vs RR171-399xxx) but subtly different in typography. 

OS: Ubuntu 18.04

"old" : libnfc.general	"ACS / ACR122U PICC Interface"
"new" : libnfc.general	"RFCARD / RF1258V603 PICC Interface"

LIBNFC_LOG_LEVEL=3 nfc-list -v output files are attached (the "new" hangs and never completes)

[nfc-list.new.log3.txt](https://github.com/nfc-tools/libnfc/files/3283396/nfc-list.new.log3.txt)
[nfc-list.old.log3.txt](https://github.com/nfc-tools/libnfc/files/3283397/nfc-list.old.log3.txt)

remytms, 2019-08-09:
Facing the same issue. Seams related to this issue #535.

poolochon, 2019-08-16:
Same issue

thekix, 2019-09-26:
Could you try this version? https://github.com/thekix/libnfc

I requested a merge with libnfc yesterday.

Regards,
kix

alexandredubois, 2019-09-26:
Hi,
I think the fix provided in https://github.com/nfc-tools/libnfc/pull/561 is a bit cleaner dont'you think ?
Regards,
Alexandre

thekix, 2019-09-30:
IMO, the patch provided at #563 is better. IMO, the alternate calls are not needed wit these devices (old and new ACR122). I tested the patch (see #564) reading, writing and Writing (sector 0) with the patch provided (#563) and with new and the old devices, it works fine. IMO, the patch removing the alternate calls is better, because removes code not needed by these devices.