nfc-tools/libnfc #548: SCL3711 fails on current master, USB timeout

philpem, 2019-07-15:
**Issue summary**: The SCL3711 fails to work with the current libnfc master, but works with the Ubuntu packaged version of libnfc (v1.7.1-4build1). 

**Git revision**: c42e2502d4627d3ea62f83c32677b100bb3cebdc

**Issue details**:

When trying to scan for cards with an SCL3711, the following errors occur:

```
$ ./utils/nfc-list 
/home/philpem/NFC/libnfc/utils/.libs/nfc-list uses libnfc libnfc-1.7.1-215-gc42e250
error	libnfc.driver.pn53x_usb	Unable to write to USB (Connection timed out)
error	libnfc.driver.pn53x_usb	Unable to write to USB (Connection timed out)
error	libnfc.driver.pn53x_usb	Unable to write to USB (Connection timed out)
nfc-list: ERROR: Unable to open NFC device: pn53x_usb:002:029
```

If I switch back to the Ubuntu release version of libnfc, it works fine:

```
c$ nfc-list
nfc-list uses libnfc 1.7.1
NFC device: SCM Micro / SCL3711-NFC&RW opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 03  44  
       UID (NFCID1): 04  0b  2e  32  a1  5c  80  
      SAK (SEL_RES): 20  
                ATS: 75  77  81  02  80  
```

doegox, 2019-07-15:
I confirm.
Culprit is 85100c0aaee728a3c254d4feed6ed5d8035bbd4a
@manu0401 can you look at it? If you really need your hardcoded values on Windows, maybe do it only for Windows platform... Still, will SCL3711 work on Windows with your hardcoded descriptors?

manu0401, 2019-07-16:
Philippe Teuwen <notifications@github.com> wrote:

> @manu0401 can you look at it? If you really need your hardcoded values on
> Windows, maybe do it only for Windows platform... Still, will SCL3711
> work on Windows with your hardcoded descriptors?

Err... I never used it on Windows.

Since the descriptors are supposed to be immutable, there is no reason
it will not work with hardcoded values.

-- 
Emmanuel Dreyfus
http://hcpnet.free.fr/pubz
manu@netbsd.org

doegox, 2019-07-16:
Ha sorry @manu0401 your commit comment was all talking about Windows...
So what should we do here? Reverse your commit or is there a better strategy to avoid breaking support for SCL3711?

manu0401, 2019-07-16:
On Tue, Jul 16, 2019 at 05:20:50AM +0000, Philippe Teuwen wrote:
> Ha sorry @manu0401 your commit comment was all talking about Windows...
> So what should we do here? Reverse your commit or is there a better strategy to avoid breaking support for SCL3711?

It is difficult to answer that question without knowing why it breaks.
Most propable reason is that the faked descriptors do not match the
real one. It would help if someone that experience the problem could
dump the descriptors to check if they match.

As an alternative, we can stop faking descriptors for SCL3711, but the 
corrupted descriptor bug will strike SCL3711 users. 

-- 
Emmanuel Dreyfus
manu@netbsd.org

doegox, 2019-07-16:
```
Bus 001 Device 114: ID 04e6:5591 SCM Microsystems, Inc. SCL3711-NFC&RW
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x04e6 SCM Microsystems, Inc.
  idProduct          0x5591 SCL3711-NFC&RW
  bcdDevice            2.07
  iManufacturer           1 SCM Micro
  iProduct                2 SCL3711-NFC&RW
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

quick test, with this patch it works again, the apparent swap of values sounds strange, I'll have to look at it later...
```diff
--- a/libnfc/drivers/pn53x_usb.c
+++ b/libnfc/drivers/pn53x_usb.c
@@ -134,7 +134,8 @@ struct pn53x_usb_supported_device {
 const struct pn53x_usb_supported_device pn53x_usb_supported_devices[] = {
   { 0x04CC, 0x0531, NXP_PN531,   "Philips / PN531", 0, 0, 0 },
   { 0x04CC, 0x2533, NXP_PN533,   "NXP / PN533", 0x04, 0x84, 40 },
-  { 0x04E6, 0x5591, SCM_SCL3711, "SCM Micro / SCL3711-NFC&RW", 0x04, 0x84, 40 },
+//  { 0x04E6, 0x5591, SCM_SCL3711, "SCM Micro / SCL3711-NFC&RW", 0x04, 0x84, 40 },
+  { 0x04E6, 0x5591, SCM_SCL3711, "SCM Micro / SCL3711-NFC&RW", 0x84, 0x04, 64 },
   { 0x04E6, 0x5594, SCM_SCL3712, "SCM Micro / SCL3712-NFC&RW", 0, 0, 0 },
   { 0x054c, 0x0193, SONY_PN531,  "Sony / PN531", 0, 0, 0 },
   { 0x1FD3, 0x0608, ASK_LOGO,    "ASK / LoGO", 0x04, 0x84, 40 },
```

manu0401, 2019-07-16:
On Tue, Jul 16, 2019 at 07:02:09AM +0000, Philippe Teuwen wrote:
> quick test, with this patch it works again, the apparent swap of values sounds strange, I'll have to look at it later...

Very suspicious indeed, especialy since other pn533 based devices
are also swapped.

-- 
Emmanuel Dreyfus
manu@netbsd.org

doegox, 2019-07-17:
See https://github.com/nfc-tools/libnfc/commit/a4af2be66c85b1846327b2ab73bb4c63c7b3b40f
@smortex or @neomilium do you have a SCL3712 to check if their endpoints info are the same as for the other ones ?

smortex, 2019-07-18:
@doegox I can't remember if I had one (SCL3711 rings a bell, 3712 not).  Unfortunately, I did not take any device when moving  :disappointed:

manu0401, 2019-07-18:
On Wed, Jul 17, 2019 at 08:32:07PM +0000, Philippe Teuwen wrote:
> @smortex or @neomilium do you have a SCL3712 to check if their endpoints info are the same as for the other ones ?

I have a SCL3712 but I need to rebuld a testing setup.

-- 
Emmanuel Dreyfus
manu@netbsd.org

doegox, 2019-07-18:
nice, when you have the chance to have a look, go ahead. We just need to enumerate it and complete the table. It's very likely to be the same values as the other ones.

neomilium, 2021-01-04:
No reply from @manu0401  since a year, feel free to open a new issue or reopen this one if relevant.

closed 2021-01-04 by neomilium