nfc-tools/libnfc #550: Pn532 RF Transmission Error" (-20)

MaxMaxov93, 2019-07-31:
Hello. I'm faced with a problem. I'm trying to use nfc-list command and a log level 3 and it throws an error:
debug	libnfc.chip.pn53x	Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20))
I'm using Kali and a new pn532. Is it a hardware issue? Thanks in advance.

Full log
```
root@kali:~# LIBNFC_LOG_LEVEL=3 nfc-list

debug	libnfc.config	key: [allow_autoscan], value: [false]

debug	libnfc.config	key: [allow_intrusive_scan], value: [false]

debug	libnfc.config	key: [device.name], value: [mypn532]

debug	libnfc.config	key: [device.connstring], value: [pn532_uart:/dev/ttyUSB0]

debug	libnfc.config	Unable to open directory: /etc/nfc/devices.d

debug	libnfc.general	log_level is set to 3

debug	libnfc.general	allow_autoscan is set to false

debug	libnfc.general	allow_intrusive_scan is set to false

debug	libnfc.general	1 device(s) defined by user

debug	libnfc.general	#0 name: "mypn532", connstring: "pn532_uart:/dev/ttyUSB0"

nfc-list uses libnfc 1.7.1

debug	libnfc.driver.pn532_uart	Attempt to open: /dev/ttyUSB0 at 115200 bauds.

debug	libnfc.bus.uart	Serial port speed requested to be set to 115200 bauds.

debug	libnfc.chip.pn53x	Diagnose

debug	libnfc.chip.pn53x	Timeout value: 500

debug	libnfc.bus.uart	TX: 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00

debug	libnfc.chip.pn53x	SAMConfiguration

debug	libnfc.chip.pn53x	Timeout value: 1000

debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 14 01 17 00

debug	libnfc.bus.uart	RX: 00 00 ff 00 ff 00

debug	libnfc.chip.pn53x	PN53x ACKed

debug	libnfc.bus.uart	RX: 00 00 ff 02 fe

debug	libnfc.bus.uart	RX: d5 15

debug	libnfc.bus.uart	RX: 16 00

debug	libnfc.bus.uart	TX: 00 00 ff 09 f7 d4 00 00 6c 69 62 6e 66 63 be 00

debug	libnfc.bus.uart	RX: 00 00 ff 00 ff 00

debug	libnfc.chip.pn53x	PN53x ACKed

debug	libnfc.bus.uart	RX: 00 00 ff 09 f7

debug	libnfc.bus.uart	RX: d5 01

debug	libnfc.bus.uart	RX: 00 6c 69 62 6e 66 63

debug	libnfc.bus.uart	RX: bc 00

[... 587 lines cut ...]

debug	libnfc.chip.pn53x	PN53x ACKed

debug	libnfc.bus.uart	RX: 00 00 ff 02 fe

debug	libnfc.bus.uart	RX: d5 33

debug	libnfc.bus.uart	RX: f8 00

debug	libnfc.chip.pn53x	PowerDown

debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 16 f0 26 00

debug	libnfc.bus.uart	RX: 00 00 ff 00 ff 00

debug	libnfc.chip.pn53x	PN53x ACKed

debug	libnfc.bus.uart	RX: 00 00 ff 03 fd

debug	libnfc.bus.uart	RX: d5 17

debug	libnfc.bus.uart	RX: 00

debug	libnfc.bus.uart	RX: 14 00
```

MaxMaxov93, 2019-07-31:
Full log
[nfclist-log.txt](https://github.com/nfc-tools/libnfc/files/3453432/nfclist-log.txt)

scottaric, 2022-02-27:
@MaxMaxov93 I'm having the same issue with a Pi 4 Model B.  Did you ever find a solution?

MaxMaxov93, 2022-02-28:
@scottaric in my case it happened to be that the device was broken. Then I bought and tested three more brand new ones (for different purposes) and everything was working correctly.

closed 2022-02-28 by MaxMaxov93

yellowzq, 2023-06-05:
Can you tell me what device are you using?

yellowzq, 2023-06-05:
I had the same problem with it.
I use the Acr122U to relay.

ld21, 2023-07-14:
I'm having the same issue (debug	libnfc.chip.pn53x	Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20))) when trying to use nfc-mfclassic w a dump.mfd using one of adafruits PN532 boards [https://www.adafruit.com/product/364](url). Is there another idea what could be wrong other than it is broken? (I can perfectly read nfc cards with the device).