nfc-tools/libnfc #558: Website not working

stigok, 2019-09-01:
The website https://nfc-tools.org has not been working for at least some weeks now.

smortex, 2019-10-29:
@doegox @neomilium do you know if Roel is still managing this domain?  (also libnfc.org redirects to this broken domain).

doegox, 2019-10-30:
no idea :(

neomilium, 2019-10-30:
@smortex it was Roel who manage libnfc.org and nfc-tools.org. I haven't be noticed of a change.

doegox, 2019-10-30:
I sent him a mail on his professional address, let's see...

peternewman, 2019-11-05:
In the short term, Web Archive has it:
http://web.archive.org/web/20190723132538/http://nfc-tools.org/index.php/Main_Page

darconeous, 2019-11-06:
Does it really make sense to have a mediawiki setup? I would think just having a static page (<https://github.com/nfc-tools/nfc-tools.github.io> perhaps?) for info and generated documentation would be sufficient, and use the built-in github wiki support instead.

doegox, 2019-11-06:
yes we should try to extract the source of wiki pages on archive.org and transfer them to the github wiki of the project

darconeous, 2019-11-06:
I'm guessing Roel is MIA?

doegox, 2019-11-06:
I didn't get any reply indeed

darconeous, 2019-11-06:
Does he control the domain registrations as well?

doegox, 2019-11-06:
yes

doegox, 2019-11-07:
this evening we got news from Roel :)
So the website is back:
http://nfc-tools.org/index.php?title=Main_Page
and in the process of getting back access to it, I fixed the forum as well:
http://forums.nfc-tools.org/

doegox, 2019-11-07:
thanks @neomilium , you had better success than I reaching Roel ;)

smortex, 2019-11-08:
Hooray!  Thanks for handling this, I think we can close this now :wink:

closed 2019-11-08 by smortex