nfc-tools/libnfc #564: Writing to a card is only partially successfull

targodan, 2019-09-28:
I recently got an ACR122 and am also affected by #535. Can confirm, pull-request #563 solves the issue. I'm currently running a self compiled version based on commit f8b28523d710c2354e1dc9094be5ebbaff494ea3 with #535 merged in.

I do not know if this is related to the library, the pull-request or my reader/writer, but I'm having trouble writing a dump to a card. It only seems to successfully write the first block of each sector. It neither writes to the other sectors nor does it overwrite the keys or access bits.
I tried this with two types of cards, both the same problem, so I doubt it's the card(s) that are faulty.

Here's what I do:

```
sudo nfc-mfclassic w a writeThisToCard.mfd dumpOfCardPreWrite.mfd
sudo nfc-mfclassic w b writeThisToCard.mfd dumpOfCardPreWrite.mfd
```

No errors come up and in the end it says `Done, 60 of 64 blocks written.` each time. (I know blocks 0-3 are locked unless its a magic card.)

My reader is a ACR122U-A9

thekix, 2019-09-30:
Hello,

no problem here using the new and de old ACR122U with patch  #535. Check that you do not have installed other libnfc in your system. I did this here:

```
export LD_LIBRARY_PATH=/home/kix/src/nfc/libnfc/libnfc/.libs/
./nfc-mfclassic r a u lector-good-step3.mfd
./nfc-mfclassic W a u lector-good-step3.mfd
./nfc-mfclassic r a u lector-good-step4.mfd
```

And the diff between lector-good-step3.mfd and lector-good-step4.mfd is ok. I tested it with "w" and "W".

Regards,
kix

targodan, 2019-09-30:
Thank you for testing. Good point. I'll double check that there is no system-lib related problem. I hadn't used the `LD_LIBRARY_PATH`, I just set the cmake prefix to `/usr`. Might be that some libs didn't get overwritten.

I'll close it for now as this is very likely to be the problem. I'll report back once I got around to testing it.

targodan, 2019-10-01:
All right, I'm back. I've cleaned up my system by uninstalling the package and manually installing the built version with the fix. Here's what I did: (I just use mfoc for dumps because I'm lazy, I cross validated the dumps using mfclassictool on my android phone, they check out)

```bash
# Starting with a clean, freshly bought card.
> mfoc -O empty.mfd
> nfc-mfclassic w a u wantedData.mfd
NFC reader: ACS / ACR122U PICC Interface opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 79  84  34  8e  
      SAK (SEL_RES): 08  
Guessing size: seems to be a 1024-byte card
Writing 64 blocks |............................................................|
Done, 60 of 64 blocks written.
> nfc-mfclassic w a u wantedData.mfd empty.mfd # just to be sure
NFC reader: ACS / ACR122U PICC Interface opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 79  84  34  8e  
      SAK (SEL_RES): 08  
Guessing size: seems to be a 1024-byte card
Writing 64 blocks |............................................................|
Done, 60 of 64 blocks written.
> mfoc -O result.mfd
# ... found all sectors use default key ffffffffff, no errors in the output
```

The `wantedData.mfd` contains some custom data for all blocks, including custom keys and access bits. I want the nfc tool to write *all* the data, including my keys and access bits. I took the access bits from a card I had lying around and verified them using mfdread, they check out and are a reasonable setup for a secure card. (well, as secure as mifare classic is, which means not at all :wink: )

Looking at the `result.mfd` dump however, only the first block of each sector is written. The keys, as well as the access bits, are untouched!

I tried this on two different card models, one of which is a magic card, the other was provided along with the reader.

I added the three files as [hex dumps to gist](https://gist.github.com/targodan/6f16e827fa726c745b88ead48fe024e0), if you want to look at it.

Here's the output `LIBNFC_LOG_LEVEL=3 nfc-mfclassic w a u wantedData.mfd`:

```
info	libnfc.config	Unable to open file: /etc/nfc/libnfc.conf
debug	libnfc.config	Unable to open directory: /etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
debug	libnfc.driver.acr122_usb	device found: Bus 002 Device 005 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:002:005"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00 
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 3b 00 
debug	libnfc.driver.acr122_usb	ACR122 PICC Operating Parameters
debug	libnfc.driver.acr122_usb	TX: 6f 05 00 00 00 00 00 00 00 00 ff 00 51 00 00 
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 90 00 
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02 
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00 
debug	libnfc.chip.pn53x	SetParameters
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 12 14 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 13 90 00 
debug	libnfc.general	"ACS / ACR122U PICC Interface" (acr122_usb:002:005) has been claimed.
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 11 00 00 00 00 00 00 00 00 ff 00 00 00 0c d4 06 63 02 63 03 63 0d 63 38 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 09 00 00 00 00 00 00 81 00 d5 07 80 80 00 08 00 90 00 
debug	libnfc.chip.pn53x	PN53X_REG_CIU_Status2 (Contain status flags of the receiver, transmitter and Data Mode Detector)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0a 00 00 00 00 00 00 00 00 ff 00 00 00 05 d4 08 63 38 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00 
[... 115 lines cut ...]
debug	libnfc.driver.acr122_usb	TX: 6f 14 00 00 00 00 00 00 00 00 ff 00 00 00 0f d4 40 01 60 38 ff ff ff ff ff ff 79 84 34 8e 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 41 00 90 00 
debug	libnfc.chip.pn53x	InDataExchange
debug	libnfc.driver.acr122_usb	TX: 6f 1a 00 00 00 00 00 00 00 00 ff 00 00 00 15 d4 40 01 a0 38 de ad be ef de ad be ef de ad be ef de ad be ef 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 41 00 90 00 
....debug	libnfc.chip.pn53x	InDataExchange
debug	libnfc.driver.acr122_usb	TX: 6f 14 00 00 00 00 00 00 00 00 ff 00 00 00 0f d4 40 01 60 3c ff ff ff ff ff ff 79 84 34 8e 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 41 00 90 00 
debug	libnfc.chip.pn53x	InDataExchange
debug	libnfc.driver.acr122_usb	TX: 6f 1a 00 00 00 00 00 00 00 00 ff 00 00 00 15 d4 40 01 a0 3c de ad be ef de ad be ef de ad be ef de ad be ef 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 41 00 90 00 
....|
Done, 60 of 64 blocks written.
debug	libnfc.driver.acr122_usb	ACR122 Abort
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02 
[... cut ...]
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 53 00 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
```

If someone could verify that the writing works completely, including keys and access bits, on their ACR122U-A9, I'll get some new cards and try again I guess. Here's the [wantedData.mfd.gz](https://github.com/nfc-tools/libnfc/files/3678055/wantedData.mfd.gz).

targodan, 2019-10-01:
I can now say with confidence that it is not the card. I found another card I once got along with an arduino nfc module. It has data on it, I once wrote with my phone (including data in blocks other than the first block of each sector) and still only the first block of each sector is written by nfc-mfclassic.

Still, this might also mean my reader is faulty rather than the library (or some system compatibility problem).

I'm running ArchLinux 64bit on a Laptop. Kernel version: 5.3.1, glibc version: 2.29, gcc version: 9.1.0

thekix, 2019-10-01:
Did you try with "W" instead of "w" in 

nfc-mfclassic W a u wantedData.mfd empty.mfd # just to be sure

And your card must support writing the sector 0

targodan, 2019-10-01:
I did with my magic card. Same result.

This problem is not (directly) related to sector 0, block 0 though. (1k memory organized in 16 sectors with each 4 blocks of 16 bytes). It only writes successfully to the first block of *each* sector. So only to every 4th block if you will. The rest of the blocks remain untouched. [Take a look at the dumps I posted in a gist.](https://gist.github.com/targodan/6f16e827fa726c745b88ead48fe024e0)

zGrEaNz, 2019-10-05:
Me too, nfc-mfclassic only work with W write, the w command doesn't write anything.

tobiabocchi, 2019-10-24:
Same issue here, using an ACR122-A9 and @thekix 's fork of libnfc I am not able to write at all; not even on every 4th block as @targodan.
this is the output of uname -a:
Linux kali 5.2.0-kali2-amd64 #1 SMP Debian 5.2.9-2kali1 (2019-08-22) x86_64 GNU/Linux
also  I do not think the cards are faulty since I am able to write them from my phone, just writing from the AC122-A9 doesn't work.
Any help is much appriciated! Thank you in advance!

Lqp1, 2019-11-19:
Same issue here with a ACR122U-A9, writes are not complete (for example, keys stay unchanged). I'm using @thekix 's fork also. Unlike @tobiabocchi I have no error using w or W.

iriseden, 2019-11-20:
Same issue for me, the problem isn't coming from the card because with MCT it works.

Do you have some news, Thanks you in advance !!!

gebi, 2020-09-04:
i've a similar problem, the last part of each block (containing the keys) is not written by nfc-mfclassic, thus all written cards have still the default key A/B (all 0xFF) and default access bits set.
It does not matter if i use `w` or `W` to write, neither produce a working card.

eg. for W and a magic card
```
# nfc-mfclassic W ab u clone2.mfb
NFC reader: ACS / ACR122U PICC Interface opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): bb  f3  76  97
      SAK (SEL_RES): 88
RATS support: no
Guessing size: seems to be a 1024-byte card
Sent bits:     50  00  57  cd
Sent bits:     40 (7 bits)
Received bits: a (4 bits)
Sent bits:     43
Received bits: 0a
Card unlocked
Writing 64 blocks |................................................................|
Done, 64 of 64 blocks written.
```

I've verified that the dump `clone2.mfb` read with R really contains the correct keys, it seems to just not write or update them.

The cards in question work fine as after writing the first block with `nfc-mfsetuid` i can write all other sector on my android phone and there it writes all blocks ok.

I've checked with a diff and it's really only the keys and ACLs missing in each block, all other parts are written properly.

I'm using debian with the following SW versions:
```
ii  libnfc-examples 1.8.0-2      amd64        Near Field Communication (NFC) examples
ii  libnfc6:amd64  1.8.0-2      amd64        Near Field Communication (NFC) library
ii  libusb-0.1-4:amd64       2:0.1.12-32  amd64        userspace USB programming library

kernel: 5.7.0-1-amd64

# nfc-scan-device
nfc-scan-device uses libnfc 1.8.0
1 NFC device(s) found:
- ACS / ACR122U PICC Interface:
    acr122_usb:001:008
```

NocTuro974, 2020-09-26:
Same problem.
I'm using 0-block writable cards, using both w and W, with and without the 'target.mfd' argument.
The command always outputs "Done, 60 of 64 blocks written." (w) or "Done, 64 of 64 blocks written." (W)

But in fact, only the first 4 blocks get written : 

> kali@kali:~$ nfc-mfclassic W a u cle_a_copier.mfd 
> NFC reader: ACS / ACR122U PICC Interface opened
> Found MIFARE Classic card:
> ISO/IEC 14443A (106 kbps) target:
>     ATQA (SENS_RES): 00  04  
>        UID (NFCID1): b9  25  8c  02  
>       SAK (SEL_RES): 08  
> RATS support: no
> Guessing size: seems to be a 1024-byte card
> Sent bits:     50  00  57  cd  
> Sent bits:     40 (7 bits)
> Received bits: a (4 bits)
> Sent bits:     43  
> Received bits: 0a  
> Card unlocked
> Writing 64 blocks |................................................................|
> Done, 64 of 64 blocks written.

And this is what I get when I read what's inside : you can see that only the first 4 blocs (0 -> 3) are written.
Thank you in advance for your time.

> kali@kali:~$ mfoc -O supposement_transforme.mfd
> Found Mifare Classic 1k tag                                                                                                                                                                                                                
> ISO/IEC 14443A (106 kbps) target:                                                                                                                                                                                                          
>     ATQA (SENS_RES): 00  04                                                                                                                                                                                                                
> * UID size: single                                                                                                                                                                                                                         
> * bit frame anticollision supported                                                                                                                                                                                                        
>        UID (NFCID1): 54  7a  ca  ba                                                                                                                                                                                                        
>       SAK (SEL_RES): 88                                                                                                                                                                                                                    
> * Not compliant with ISO/IEC 14443-4                                                                                                                                                                                                       
> * Not compliant with ISO/IEC 18092                                                                                                                                                                                                         
>                                                                                                                                                                                                                                            
> Fingerprinting based on MIFARE type Identification Procedure:                                                                                                                                                                              
> * SmartMX with MIFARE 1K emulation                                                                                                                                                                                                         
> Other possible matches based on ATQA & SAK values:                                                                                                                                                                                         
> * Mifare Classic 1K Infineon                                                                                                                                                                                                               
>                                                                                                                                                                                                                                            
> Try to authenticate to all sectors with default keys...
> Symbols: '.' no key found, '/' A key found, '\' B key found, 'x' both keys found
> [Key: ffffffffffff] -> [.xxxxxxxxxxxxxxx]
> [Key: a0a1a2a3a4a5] -> [.xxxxxxxxxxxxxxx]
> [Key: d3f7d3f7d3f7] -> [.xxxxxxxxxxxxxxx]
> [Key: 000000000000] -> [.xxxxxxxxxxxxxxx]
> [Key: b0b1b2b3b4b5] -> [.xxxxxxxxxxxxxxx]
> [Key: 4d3a99c351dd] -> [.xxxxxxxxxxxxxxx]
> [Key: 1a982c7e459a] -> [.xxxxxxxxxxxxxxx]
> [Key: aabbccddeeff] -> [.xxxxxxxxxxxxxxx]
> [Key: 714c5c886e97] -> [.xxxxxxxxxxxxxxx]
> [Key: 587ee5f9350f] -> [.xxxxxxxxxxxxxxx]
> [Key: a0478cc39091] -> [.xxxxxxxxxxxxxxx]
> [Key: 533cb6c723f6] -> [.xxxxxxxxxxxxxxx]
> [Key: 8fd0a4f256e9] -> [.xxxxxxxxxxxxxxx]
> 
> Sector 00 - Unknown Key A               Unknown Key B
> Sector 01 - Found   Key A: ffffffffffff Found   Key B: ffffffffffff
[... cut ...]
> Block 04, type A, key ffffffffffff :00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  
> Block 03, type A, key 8829da9daf76 :00  00  00  00  00  00  7f  07  88  00  00  00  00  00  00  00  
> Block 02, type A, key 8829da9daf76 :00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  
> Block 01, type A, key 8829da9daf76 :01  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  
> Block 00, type A, key 8829da9daf76 :54  7a  ca  ba  5e  88  04  00  47  c1  1e  67  51  00  02  06
>

alan-maneg, 2020-10-03:
I have exactly the same problem. Curiously on my Raspberry all the blocks are written but not on my Dell. I don't understand this issue. May be a future update ?

targodan, 2020-10-03:
> Curiously on my Raspberry all the blocks are written

That sounds interesting. @alan-maneg could you maybe check and post the versions (and ideally commit hash if self-compiled) of the following libs/packages/stuff for both your raspy and your dell?
- Linux Kernel
- pcsc
- acr122_pscs / acr122_usb (as mentioned in the [README.md under requirements](https://github.com/nfc-tools/libnfc#requirements) )
- libnfc

I don't have a raspy running at the moment, so it might take some time, but I'll try to verify this once I have my raspy running again.

philou66, 2020-10-05:
Hi, same issue as you all for me (unable to write UID of a block0 rewritable card), on a Raspberry Pi. Raspbian OS, kernel 4.19.97-v7+, ACR122U-A9 reader.
I have tried this fork from @sam1902 that solves this other issue I had, https://github.com/nfc-tools/libnfc/issues/570 the problem remains.
I can dump full traces if it helps ?

targodan, 2021-01-20:
Had another poke at this. I believe its either a problem in the ACR122U-A9 firmware or the driver for it. Because:

I hacked my ACR122 because I got sick of these issues. I opened up the reader, cut the connections between the pn532 and the USB chip (they scrubbed the package so I have no part number I could give you). Then I bodged in the IO selection pins to select HSU (high speed UART) (I0 = 0 and I1 = 0) and soldered on some wires to exposed test pads for the serial IO. These I connected to an FTDI USB to UART converter. Now I can successfully write to the entire card (including sector 0, because magic card).

That way I can at least guarantee that the RF hardware and the pn532 is not faulty. Of course I can still not guarantee that the USB chip isn't faulty in hard- or firmware.

For additional hardware information: On the board was printed "REV 1.14C", the pn532 had the following writing on it (not sure which is the part number)
```
5321 06
CTW145
    06
TSD8381
```

On the left side of the chip where strange symbols, I presume a vendor logo. I could provide you with fotos if you're interested.

LeoBenoist, 2021-02-11:
It seem to have the same problem on a new ACR122U-A9. What is weird it that it worked with a really old 0 writable card I had but with the new cards (provided with the ACR) it only seem to copy half the blocks. The uuid is ok but the SAK is false (only with new card). I can correct the SAK with nfc-mfsetuid but it doesnt fix the partially writtend blocks. If I look at the badly writtend card it look like all key after block 4 are not written.

rrifi, 2021-02-15:
I also use acr122u
i think the problem comes from libnfc6 

because when i was using libnfc5, it worked well, i start my script and copies was automatic without error, 

now i unfortunately did an update and it doesnt work...

So I use PCSC Mifare softwares on Windows but it takes very long time because I have to copy paste each bytes manually this is boring

LeoBenoist, 2021-02-15:
So for me it was not working on Kali with the 1.8. I juste installed a Buster Debian with version 1.7 and it is working like a charm.

rrifi, 2021-02-17:
> So for me it was not working on Kali with the 1.8. I juste installed a Buster Debian with version 1.7 and it is working like a charm.

Merci t'es un boss, ça a fonctionné pour moi aussi j'ai bien fait de poster ici
---------------------------

As LeoBenoist said I just installed Debian (last version) and it's working now

dont forget to do a "sudo modprobe -r pn533 nfc" to blacklist unwanted drivers

gebi, 2021-02-18:
@rrifi could you post some more information which version you of debian you tried and worked :) ?

what was the "last version" (you meant latest?) that worked for you from debian?

debian release | libnfc | version
------------ | ----- | --------
stable / buster | libnfc5 | 1.7.1-4+b1
bullseye / testing | libnfc6 | 1.8.0-2
unstable / sid | libnfc6 | 1.8.0-2

just to be sure here

LeoBenoist, 2021-02-18:
@gebi like I said :) buster

rrifi, 2021-02-19:
@gebi  
here is the link 
https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/
i downloaded "debian-10.8.0-amd64-netinst.iso"

------------------------------------------------------
if you want, here is my Oracle VM Virtualbox exported machine (with mfoc and libnfc installed and ready to use)
https://mega.nz/file/8o9FjKJQ#VnjxLO-59ARyzunQ5II57kPKND04kmSr2Tw64a3UD-A

user debian:debian
& root:toor

gebi, 2021-02-19:
awesome thx!

rixnco, 2021-06-10:
Hi,
Any update on this problem ? I'm in the same situation as @targodan. I've opened the reader and I have the same board revision 1.14C.
Some components have not been populated. Might be the origin of the pb? Can someone with a working unit send pictures of the internal board ?

I can confirm that debian 10.9 with libnfc1.7.1 works (full writing capabilities).

matthavener, 2021-07-02:
I was seeing this issue as well on Kali Linux and an ACR122U-A9. Using debian 10.10 and 10.8, I was getting "Connection timed out" errors when running `nfc-list`. I finally got it working by building libnfc-1.8.0 (released tar.bz2) from source on a debian 10.8 live image.

thekix, 2021-07-03:
So, the problem seems to be in the Kali/distro package, not in the upstream source code. IMO, this bug could be closed.

targodan, 2021-07-03:
Doubtful. When I opened the issue i was running ArchLinux 64 bit with Kernel version 5.3.1. Also someone tried it with Raspbian (see above).

LeoBenoist, 2021-07-04:
Yes it is on other distro!

pierpierre, 2021-10-31:
Same issue here. Been trying lots of different ways, always the same result. I even tried the Debian VM image posted by @LeoBenoist but it won't see my ACR122U-A9 (beginning to wonder if there is simply a bad series of those?).
Are we the only one? Tempted to buy another one just to see, and return it in case...

CDFN, 2021-11-01:
In my case building from source from master helped. Reading and writing works flawlessly.

pierpierre, 2021-11-01:
@CDFN Did you start from the latest version? Or you built from 1.7.1?

CDFN, 2021-11-01:
latest version, you must clone repository from master branch and build from there.

Lqp1, 2021-11-01:
I confirm it worked when compiling from master on my `acr122u`; thanks!

1.7.1 was not working (but slightly different errors).

pierpierre, 2021-11-02:
Thanks guys!
I did as suggested - spent too much time doing it as my compiling knowledge is pretty basic. But it worked! Copied my building fob and it works perfectly.

bjoyal375, 2021-11-16:
🙄

IshFx, 2021-12-19:
Any updates on this ? I was facing the same issue and had to downgrade to 1.7.1 to make it work.

Does anybody know why the 1.8 version doesn't work while the 1.7.1 does work ?

pierpierre, 2021-12-19:
It works, it's just the one on Kali repo and some other repo is crap. If you compile from scratch yourself, it works. That's what I ended up doing, with my very limited knowledge of Linux, and it now works.

IshFx, 2021-12-20:
I'm on arch, and it doesn't work either ...
here is the PKGBUILD https://github.com/archlinux/svntogit-community/blob/packages/libnfc/trunk/PKGBUILD

As you can see, it downloads the binary from the github releases. So the problem is not from Kali.

CDFN, 2021-12-20:
Because github releases has incorrect package. You have to git clone master branch and start from there :smile:

vuongdat8007, 2022-03-03:
> @gebi here is the link https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/ i downloaded "debian-10.8.0-amd64-netinst.iso"
>
> if you want, here is my Oracle VM Virtualbox exported machine (with [...]

Hello,

User and password did not work.
Can you recheck please?

Thanks a lot!

rrifi, 2022-03-03:
user : debian
pass : debian

& for root
user : root
pass : toor

Odroid13, 2023-09-21:
> @gebivoici le lien https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/j'ai téléchargé "debian-10.8.0-amd64-netinst.iso"
> 
> si vous le souhaitez, voici ma machine exportée Oracle VM Virtualbox (avec mfoc et libnfc installés et prêts à l'emploi) https://mega.nz/file/8o9FjKJQ#VnjxLO-59ARyzunQ5II57kPKND04kmSr2Tw64a3UD-A
> 
> utilisateur debian:debian& root:Thank you for your virtual machine, I will leave it as soon as I have time.

Can you tell me if the commands below are correct to clone an RFID card?

To extract:
```
mfoc -P 500 -O original.mfd
mfoc -P 500 -O blank.mfd
```

To write on the blank card:
`nfc-mfclassic W a u original.mfd virgin.mfd f`

Thanks

Odroid13, 2023-09-21:
> > > @gebivoici le lien https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/j'ai téléchargé "debian-10.8.0-amd64-netinst.iso"
> > > si vous le souhaitez, voici ma machine exportée Oracle VM Virtualbox (avec mfoc et libnfc installés et prêts à l'emploi) https://mega.nz/file/8o9FjKJQ#VnjxLO-59ARyzunQ5II57kPKND04kmSr2Tw64a3UD-A
> > > utilisateur debian:debian& root:Thank you for your virtual machine, I will leave it as soon as I have time.
> > 
> > 
> > Can you tell me if the commands below are correct to clone an RFID card?
> > To extract:
> > ```
> > mfoc -P 500 -O original.mfd
> > mfoc -P 500 -O blank.mfd
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > To write on the blank card: `nfc-mfclassic W a u original.mfd blank.mfd f`
> > Thanks
> 
> Yes your commands seems to be correct
> 
> Here is how i do to extract :
> 
> mfoc -O extracted.dump or nfc-mfclassic r a extracted.dump
> 
> To write :
> 
> nfc-mfclassic W a dataToCopy.dump
> 
> (i just specify the dump which contains the data i want to write in the blank tag)

Thank you for your virtual machine, I will leave it as soon as I have time.

Thank you for your response ^^

I just did a test with your debian VM.

I extracted my original badge
Same for the blank card.

When I want to write on the blank card with the command:

  nfc-mfclassic W a u original.mfd blank.mfd f
Or
sudo nfc-mfclassic W a original.dmp blank.dmp

He tells me as a result

```
uessing size: seems to be a 1024-byte card
Sent bits:     50  00  57  cd  
Sent bits:     40 (7 bits)
unlock failure!
```

How do I know if the blank card I have is with block O unlock

sorry I'm a beginner

rrifi, 2023-09-21:
> Thank you for your virtual machine, I will leave it as soon as I have time.
>
> Thank you for your response ^^
>
> I just did a test with your debian VM.
>
> I extracted my original badge Same for the blank [...]

This error appears when you can't write in the 0 block

Is your tag UID Changeable ? 
 
If not, try with W but lower caps

rrifi, 2023-09-21:
> I tried with Kali and I also got an error, I think my blank card is not writing to block 0. Is there a trick to know if it's a good card or not? THANKS [image]
> 
> `with capital W command`
> 
> [image]

Maybe it's an old badge that you found, and no longer had a use for, so you decided to write on it? 
Or is it a badge that you specifically bought on Aliexpress for the purpose of writing on it?

Because if you use a tag which is not made to write data on it, you may meet some struggles (because some blocks can be write protected, etc. )

You should use Mifare 1K block 0 Changeable from Aliexpress
--------------------------------------------------

The trick to know if uid is changeable or not :

Take all the content of https://gist.githubusercontent.com/hiviah/199362d9b60e30e98ce06ad971d761a8/raw/05ee247cc5f7301e0de7788c7589fdf4543a847f/nfc-detect-rewritable-uid-mifare.c

Save it in a file, like "detect.c"

Compile it -> run the command "gcc detect.c -o detect -lnfc" 
(in lnfc, it's a L not a i)

Put your tag on the ACR122U, then run the program by launching the command "./detect" (which is the compiled program)

Odroid13, 2023-09-21:
Thanks , 
I bought the device:
ACR122U and in the box there were 5 white cards supplied with the reader.

I made a backup of my wash card and would like to write on a blank card.

I will try your advice to check if blank cards have block 0.

If in doubt, I will order other cards on Aliexpress but I have doubts because it says Android MCT compatible:

I just found this amazing item on AliExpress. Take a look there ! €2.02 22% off | 5pcs/lot 13.56Mhz RFID ID Card Sticker Keyfob Tag Changeable Night 0ampa NDavid Android MCT Copy Clone Duplicator
https://a.aliexpress.com/_ExCXLsj

Odroid13, 2023-09-21:
Feedback

[image]

😭

Octavian0073, 2024-09-06:
I've checked my chinese magic cards w/ the script, and the feedback validates them, but for whatever reason it s not possible to write the uid. Getting the same:

 "Unlock command [1/2]: failed / not acknowledged." 

Can anyone please help?

StarScream159, 2025-03-15:
@Octavian0073, I have the same problem. detect reports success. Writing fails with same error.

Octavian0073, 2025-03-17:
@StarScream159 yes. Check one of my repos, i have published one called mifare. There are some c programs that are modified to be able to write and entire card or only to change the uid.

tiene77, 2025-03-23:
> [@gebi](https://github.com/gebi) here is the link https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/ i downloaded "debian-10.8.0-amd64-netinst.iso"
>
> if you want, here is my Oracle VM [...]

Hi , i have the same error with kali.
I download your debian vm, but i’ m a biginner and i’m lost, can i have some help ? Pls
Thanks

rrifi, 2025-03-23:
Be sure you have installed Virtualbox (I use it on Windows) its very easy
to use

Then just launch the dowloaded file, it will automatically save the VM in
Virtualbox

Then open and connect to the VM with login 'debian' and password 'debian'

tiene77, 2025-03-23:
> Be sure you have installed Virtualbox (I use it on Windows) its very easy
> to use
> 
> Then just launch the dowloaded file, it will automatically save the VM in
> Virtualbox
> 
> Then open and connect to the VM with login 'debian' and password 'debian'
> 
> […](#)
> 
>  [@gebi](https://github.com/gebi) <https://github.com/gebi> here is the link
>  https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/ i downloaded
>  "debian-10.8.0-amd64-netinst.iso"
> 
>  if you want, here is my Oracle VM Virtualbox exported machine (with mfoc
>  and libnfc installed and ready to use)
>  https://mega.nz/file/8o9FjKJQ#VnjxLO-59ARyzunQ5II57kPKND04kmSr2Tw64a3UD-A
> 
>  user debian:debian & root:toor
> 
>  Hi , i have the same error with kali.
>  I download your debian vm, but i’ m a biginner and i’m lost, can i have
>  some help ? Pls
>  Thanks
> 
>  <[#564 (comment)](https://github.com/nfc-tools/libnfc/issues/564#issuecomment-2746364849)>,
>  <[#564 (comment)](https://github.com/nfc-tools/libnfc/issues/564#issuecomment-2746364849)>
> 
>  [@gebi](https://github.com/gebi) <https://github.com/gebi> here is the link
>  https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/ i downloaded
>  "debian-10.8.0-amd64-netinst.iso"
> 
>  if you want, here is my Oracle VM Virtualbox exported machine (with mfoc
>  and libnfc installed and ready to use)
>  https://mega.nz/file/8o9FjKJQ#VnjxLO-59ARyzunQ5II57kPKND04kmSr2Tw64a3UD-A
> 
>  user debian:debian & root:toor
> 
>  Hi , i have the same error with kali.
>  I download your debian vm, but i’ m a biginner and i’m lost, can i have
>  some help ? Pls
>  Thanks
> 
>  <[#564 (comment)](https://github.com/nfc-tools/libnfc/issues/564#issuecomment-2746364849)>,

Thank you
Yes i have vmware for mac , the vm works debian is open ,  but the command are différent than kali ?
Also are you French ? Me yes

rrifi, 2025-03-23:
Yep i'm french

Yes commands are the same,
Where are you stuck ?

tiene77, 2025-03-23:
Cool

Nfc-list doesn’t work because no nfc device found.
So i write  « sudo modprobe -r pn533 nfc pn533_usb »
Password is asked , i put it and i have the message « debian n’apparaît pas dans le fichier sudoers. Cet événement sera signalé »
> Yep i'm french
> 
> Yes commands are the same,
> Where are you stuck ?
> 
> Le dim. 23 mars 2025 à 20:28, tiene77 ***@***.***> a écrit :
> […](#)

rrifi, 2025-03-23:
Maybe it's struggling to find the usb because of the USB Passtrhough (since
you're on a Mac) idk

But this message is caused because the user 'debian' is not admin by default

Try this to add it to the admins :
> su -
(the password is of root 'toor')
> usermod -aG sudo debian

Then reboot the debian machine and retry the
> sudo modprobe -r pn533 nfc pn533_usb

tiene77, 2025-03-23:
NFC device still not found, i try on Windows pc (i use kali on this one ) and nfc device not found also, ( the light is green on my acr122u)

> Maybe it's struggling to find the usb because of the USB Passtrhough (since
> you're on a Mac) idk
>
> But this message is caused because the user 'debian' is not admin by default
>
> Try this to add it to [...]

rrifi, 2025-03-23:
Bro did you connect the acru122u to your virtual machine ?

You have to connect it through the USB icon, en bas a droite de la fenetre

tiene77, 2025-03-23:
Yes it’s connected, i said device is ok in my message , i mean it works sorry.
 With the command nfc-mfclassic W a mondumpacopier.mfd i have unlock failure at the end

> Bro did you connect the acru122u to your virtual machine ?
> 
> You have to connect it through the USB icon, en bas a droite de la fenetre
> 
> Le lun. 24 mars 2025, 00:04, tiene77 ***@***.***> a écrit :
> […](#)

[2 more replies]