nfc-tools/libnfc #566: Failure to write to data block 4

fxcoudert, 2019-10-09:
I am using `libnfc` master (with patch from https://github.com/nfc-tools/libnfc/pull/561), on an ACS / ACR122U reader, and trying to write to a card with rewritable UID. The write fails with:

```
$ nfc-mfclassic W a dump.fx.9ed9be0d nom_badge_vierge.dmp
NFC reader: ACS / ACR122U PICC Interface opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): d6  75  8d  29  
      SAK (SEL_RES): 08  
Guessing size: seems to be a 1024-byte card
Sent bits:     50  00  57  cd  
Sent bits:     40 (7 bits)
Warning: Unlock command [1/2]: failed / not acknowledged.
Writing 64 blocks |Failure to write to data block 4
x
```

The cards I am trying to write to are these with rewritable UID, I believe they are gen B / second generation: https://www.amazon.fr/Lot-badges-Rfid-Mif-13-56Mhz/dp/B07GD5BQ1T 

Verbose output:

```
$ LIBNFC_LOG_LEVEL=3 nfc-mfclassic W a dump.fx.9ed9be0d nom_badge_vierge.dmp
debug	libnfc.config	Parse error on line #1: allow_intrusive_scan=yes
debug	libnfc.config	Unable to open directory: /usr/local/Cellar/libnfc/HEAD-f8b2852/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
debug	libnfc.driver.acr122_usb	device found: Bus 020 Device 006 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:020:006"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00 
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 3b 00 
debug	libnfc.driver.acr122_usb	ACR122 PICC Operating Parameters
debug	libnfc.driver.acr122_usb	TX: 6f 05 00 00 00 00 00 00 00 00 ff 00 51 00 00 
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 90 00 
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02 
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00 
debug	libnfc.chip.pn53x	SetParameters
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 12 14 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 13 90 00 
debug	libnfc.general	"ACS / ACR122U PICC Interface" (acr122_usb:020:006) has been claimed.
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 11 00 00 00 00 00 00 00 00 ff 00 00 00 0c d4 06 63 02 63 03 63 0d 63 38 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 09 00 00 00 00 00 00 81 00 d5 07 80 80 00 00 00 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 01 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 ff ff ff 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 13 00 00 00 00 00 00 00 00 ff 00 00 00 0e d4 06 63 02 63 03 63 05 63 38 63 3c 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 0a 00 00 00 00 00 00 81 00 d5 07 80 80 40 00 10 00 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 00 01 02 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	SetParameters
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 12 04 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 13 90 00 
NFC reader: ACS / ACR122U PICC Interface opened
debug	libnfc.chip.pn53x	InListPassiveTarget
debug	libnfc.chip.pn53x	Timeout value: 300
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 4a 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 0e 00 00 00 00 00 00 81 00 d5 4b 01 01 00 04 08 04 d6 75 8d 29 90 00 
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): d6  75  8d  29  
      SAK (SEL_RES): 08  
debug	libnfc.chip.pn53x	InCommunicateThru
debug	libnfc.chip.pn53x	No timeout
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 42 e0 50 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 43 02 90 00 
debug	libnfc.chip.pn53x	Chip error: "CRC Error" (02), returned error: "RF Transmission Error" (-20))
debug	libnfc.chip.pn53x	InListPassiveTarget
debug	libnfc.chip.pn53x	Timeout value: 300
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 4a 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 0e 00 00 00 00 00 00 81 00 d5 4b 01 01 00 04 08 04 d6 75 8d 29 90 00 
Guessing size: seems to be a 1024-byte card
Sent bits:     50  00  57  cd  
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 06 63 02 63 03 
debug	libnfc.driver.acr122_usb	RX: 80 06 00 00 00 00 00 00 81 00 d5 07 80 80 90 00 
debug	libnfc.chip.pn53x	PN53X_REG_CIU_TxMode (Defines the transmission data rate and framing during transmission)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_RxMode (Defines the transmission data rate and framing during receiving)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0d 00 00 00 00 00 00 00 00 ff 00 00 00 08 d4 08 63 02 00 63 03 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00 
debug	libnfc.chip.pn53x	InCommunicateThru
debug	libnfc.chip.pn53x	No timeout
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 42 50 00 57 cd 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 43 01 90 00 
debug	libnfc.chip.pn53x	Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20))
Sent bits:     40 (7 bits)
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 06 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 07 00 90 00 
debug	libnfc.chip.pn53x	PN53X_REG_CIU_BitFraming (Adjustments for bit oriented frames)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0a 00 00 00 00 00 00 00 00 ff 00 00 00 05 d4 08 63 3d 07 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00 
debug	libnfc.chip.pn53x	InCommunicateThru
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 42 40 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 43 01 90 00 
debug	libnfc.chip.pn53x	Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20))
Warning: Unlock command [1/2]: failed / not acknowledged.
Writing 64 blocks |debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0d 00 00 00 00 00 00 00 00 ff 00 00 00 08 d4 06 63 02 63 03 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 07 00 00 00 00 00 00 81 00 d5 07 00 00 07 90 00 
debug	libnfc.chip.pn53x	PN53X_REG_CIU_TxMode (Defines the transmission data rate and framing during transmission)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_RxMode (Defines the transmission data rate and framing during receiving)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_BitFraming (Adjustments for bit oriented frames)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 10 00 00 00 00 00 00 00 00 ff 00 00 00 0b d4 08 63 02 80 63 03 80 63 3d 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00 
debug	libnfc.chip.pn53x	InDataExchange
debug	libnfc.driver.acr122_usb	TX: 6f 1a 00 00 00 00 00 00 00 00 ff 00 00 00 15 d4 40 01 a0 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 41 01 90 00 
debug	libnfc.chip.pn53x	Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20))
Failure to write to data block 4
debug	libnfc.driver.acr122_usb	ACR122 Abort
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02 
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00 
debug	libnfc.chip.pn53x	InRelease
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 52 00 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 53 00 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
```

fxcoudert, 2019-10-09:
Starting with the unused card, the complete sequence of events is:

```
$ nfc-list                                                                 
nfc-list uses libnfc 1.7.1
NFC device: ACS / ACR122U PICC Interface opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): d6  1d  1d  29  
      SAK (SEL_RES): 08  
$ mfoc -P 500 -O dump.clean          
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
* UID size: single
* bit frame anticollision supported
       UID (NFCID1): d6  1d  1d  29  
      SAK (SEL_RES): 08  
* Not compliant with ISO/IEC 14443-4
* Not compliant with ISO/IEC 18092

Fingerprinting based on MIFARE type Identification Procedure:
* MIFARE Classic 1K
* MIFARE Plus (4 Byte UID or 4 Byte RID) 2K, Security level 1
* SmartMX with MIFARE 1K emulation
Other possible matches based on ATQA & SAK values:

Try to authenticate to all sectors with default keys...
Symbols: '.' no key found, '/' A key found, '\' B key found, 'x' both keys found
[Key: ffffffffffff] -> [xxxxxxxxxxxxxxxx]
[Key: a0a1a2a3a4a5] -> [xxxxxxxxxxxxxxxx]
[Key: d3f7d3f7d3f7] -> [xxxxxxxxxxxxxxxx]
[Key: 000000000000] -> [xxxxxxxxxxxxxxxx]
[Key: b0b1b2b3b4b5] -> [xxxxxxxxxxxxxxxx]
[Key: 4d3a99c351dd] -> [xxxxxxxxxxxxxxxx]
[Key: 1a982c7e459a] -> [xxxxxxxxxxxxxxxx]
[Key: aabbccddeeff] -> [xxxxxxxxxxxxxxxx]
[Key: 714c5c886e97] -> [xxxxxxxxxxxxxxxx]
[Key: 587ee5f9350f] -> [xxxxxxxxxxxxxxxx]
[Key: a0478cc39091] -> [xxxxxxxxxxxxxxxx]
[Key: 533cb6c723f6] -> [xxxxxxxxxxxxxxxx]
[Key: 8fd0a4f256e9] -> [xxxxxxxxxxxxxxxx]

Sector 00 -  FOUND_KEY   [A]  Sector 00 -  FOUND_KEY   [B]  
Sector 01 -  FOUND_KEY   [A]  Sector 01 -  FOUND_KEY   [B]  
Sector 02 -  FOUND_KEY   [A]  Sector 02 -  FOUND_KEY   [B]  
Sector 03 -  FOUND_KEY   [A]  Sector 03 -  FOUND_KEY   [B]  
Sector 04 -  FOUND_KEY   [A]  Sector 04 -  FOUND_KEY   [B]  
Sector 05 -  FOUND_KEY   [A]  Sector 05 -  FOUND_KEY   [B]  
Sector 06 -  FOUND_KEY   [A]  Sector 06 -  FOUND_KEY   [B]  
Sector 07 -  FOUND_KEY   [A]  Sector 07 -  FOUND_KEY   [B]  
Sector 08 -  FOUND_KEY   [A]  Sector 08 -  FOUND_KEY   [B]  
Sector 09 -  FOUND_KEY   [A]  Sector 09 -  FOUND_KEY   [B]  
Sector 10 -  FOUND_KEY   [A]  Sector 10 -  FOUND_KEY   [B]  
Sector 11 -  FOUND_KEY   [A]  Sector 11 -  FOUND_KEY   [B]  
Sector 12 -  FOUND_KEY   [A]  Sector 12 -  FOUND_KEY   [B]  
Sector 13 -  FOUND_KEY   [A]  Sector 13 -  FOUND_KEY   [B]  
Sector 14 -  FOUND_KEY   [A]  Sector 14 -  FOUND_KEY   [B]  
Sector 15 -  FOUND_KEY   [A]  Sector 15 -  FOUND_KEY   [B]  

We have all sectors encrypted with the default keys..

Auth with all sectors succeeded, dumping keys to a file!
Block 63, type A, key ffffffffffff :00  00  00  00  00  00  ff  07  80  69  ff  ff  ff  ff  ff  ff  
Block 62, type A, key ffffffffffff :00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  
Block 61, type A, key ffffffffffff :00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  
Block 60, type A, key ffffffffffff :00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  
Block 59, type A, key ffffffffffff :00  00  00  00  00  00  ff  07  80  69  ff  ff  ff  ff  ff  ff  
Block 58, type A, key ffffffffffff :00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  
Block 57, type A, key ffffffffffff :00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  
[... 66 lines cut ...]
Sent bits:     50  00  57  cd  
Sent bits:     40 (7 bits)
Warning: Unlock command [1/2]: failed / not acknowledged.
Writing 64 blocks |Failure to write to data block 4
x
```

Note that writing without UID (`w`) does not fail:

```
$ nfc-mfclassic w a dump.fx.9ed9be0d dump.clean
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
nfc-mfclassic: ERROR: Error opening NFC reader
rmeur ~/Desktop/RFID $ sudo killall -9 com.apple.ifdreader          
rmeur ~/Desktop/RFID $ nfc-mfclassic w a dump.fx.9ed9be0d dump.clean
NFC reader: ACS / ACR122U PICC Interface opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): d6  1d  1d  29  
      SAK (SEL_RES): 08  
Guessing size: seems to be a 1024-byte card
Writing 64 blocks |............................................................|
Done, 60 of 64 blocks written.
```

fxcoudert, 2019-10-09:
poking @quantum-x, if I understand well the code involved is theirs

crazy-projects, 2019-11-04:
Same problem here, can't write gen 2 CUID magic card (the ones with the block 0 directly writable, and no magic command) with same error message. 

However I can write "normally" (lowercase w) to the tag, and of course, the block 0 remains untouched.

And the card works with and Android phone and MIFARE Classic Tool app.

sgadrat, 2020-01-31:
I have the exact same tags that the ones linked by @fxcoudert, and the same problem.

I bypassed it by forcing `magic2` to `true` in nfc-mfclassic.c then recompiling. So it seems that the magic tag detection fails to recognize it correctly.

Here is my patch:
```diff
diff --git a/utils/nfc-mfclassic.c b/utils/nfc-mfclassic.c
index ba07b6f..8b29b65 100644
--- a/utils/nfc-mfclassic.c
+++ b/utils/nfc-mfclassic.c
@@ -70,7 +70,7 @@ static bool bUseKeyFile;
 static bool bForceKeyFile;
 static bool bTolerateFailures;
 static bool bFormatCard;
-static bool magic2 = false;
+static bool magic2 = true;
 static bool magic3 = false;
 static bool unlocked = false;
 static bool bForceSizeMismatch;
```

(While it works around the problem, it is not a solution. It will break compatibility with gen1 magic tags, which will not be unlocked, and normal mifare tags, wich will fail to write block 0)

Hacking a little more the source to get some info around the magic detection, it seems that `get_rats()` returns `-20`, forbidding any subsequent process. Here is the output with a printf of get_rats() return code (search for "RATS" in the log:

```
info	libnfc.config	Unable to open file: /home/sylvain/apps/prefix/etc/nfc/libnfc.conf
debug	libnfc.config	Unable to open directory: /home/sylvain/apps/prefix/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
debug	libnfc.driver.acr122_usb	device found: Bus 001 Device 006 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:001:006"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00 
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 3b 00 
debug	libnfc.driver.acr122_usb	ACR122 PICC Operating Parameters
debug	libnfc.driver.acr122_usb	TX: 6f 05 00 00 00 00 00 00 00 00 ff 00 51 00 00 
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 90 00 
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02 
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00 
debug	libnfc.chip.pn53x	SetParameters
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 12 14 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 13 90 00 
debug	libnfc.general	"ACS / ACR122U PICC Interface" (acr122_usb:001:006) has been claimed.
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 11 00 00 00 00 00 00 00 00 ff 00 00 00 0c d4 06 63 02 63 03 63 0d 63 38 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 09 00 00 00 00 00 00 81 00 d5 07 80 80 00 08 00 90 00 
debug	libnfc.chip.pn53x	PN53X_REG_CIU_Status2 (Contain status flags of the receiver, transmitter and Data Mode Detector)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0a 00 00 00 00 00 00 00 00 ff 00 00 00 05 d4 08 63 38 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 01 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 ff ff ff 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 13 00 00 00 00 00 00 00 00 ff 00 00 00 0e d4 06 63 02 63 03 63 05 63 38 63 3c 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 0a 00 00 00 00 00 00 81 00 d5 07 80 80 40 00 10 00 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 00 01 02 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	SetParameters
[... 9 lines cut ...]
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 43 02 90 00 
debug	libnfc.chip.pn53x	Chip error: "CRC Error" (02), returned error: "RF Transmission Error" (-20))
debug	libnfc.chip.pn53x	InListPassiveTarget
debug	libnfc.chip.pn53x	Timeout value: 300
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 4a 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 0e 00 00 00 00 00 00 81 00 d5 4b 01 01 00 04 08 04 2a 78 23 18 90 00 
NFC reader: ACS / ACR122U PICC Interface opened
Expected MIFARE Classic card with UID starting as: 6504c12a
Got card with UID starting as:                     2a782318
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 2a  78  23  18  
      SAK (SEL_RES): 08  
RATS: failed, res -20
Guessing size: seems to be a 1024-byte card
Reading out 64 blocks |debug	libnfc.chip.pn53x	InDataExchange
debug	libnfc.driver.acr122_usb	TX: 6f 14 00 00 00 00 00 00 00 00 ff 00 00 00 0f d4 40 01 60 3f ff ff ff ff ff ff 2a 78 23 18 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 41 00 90 00 
debug	libnfc.chip.pn53x	InDataExchange

[Truncated, there is lots of TX/RX, it reads the whole tag]
```

I gone far beyond my understanding of the subject. From now, I'll let people who have a clue of what all this mean do their magic. Hope it helps. Thank you for maintaining libnfc!

vkravets, 2021-10-13:
I've faced with the same issue and it seems it tries to write gen3 fob, which new family of the fobs. With the master build, some of related commits to gen3 is not at 1.8.0. So master build resolve this issue.

See https://github.com/nfc-tools/libnfc/pull/608

@fxcoudert try to build from master and try to write again

ilyesAj, 2023-10-18:
@vkravets i confirm that the issue is resolved when built from master branch.
@neomilium is it possible to release a new version of libnfc ? it will avoid building from master .

tony1016, 2024-03-02:
interesting.I've faced the same problem.Then I write with -f once,then write normally.
```
tl@alpine-on-gk41 ~/P/l/utils (master) [1]> sudo ./nfc-mfclassic W a u ~/Downloads/apartment.card
NFC reader: microBuilder.eu opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 22  c7  eb  0d  
      SAK (SEL_RES): 08  
RATS support: no
Guessing size: seems to be a 1024-byte card
Sent bits:     50  00  57  cd  
Sent bits:     40 (7 bits)
Warning: Unlock command [1/2]: failed / not acknowledged.
Trying to rewrite block 0 on a direct write tag.
Writing 64 blocks |....!
Error: authentication failed for block 04
tl@alpine-on-gk41 ~/P/l/utils (master) [1]> sudo ./nfc-mfclassic f W a ~/Downloads/apartment.card ~/Downloads/apartment.card
NFC reader: microBuilder.eu opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 22  c7  eb  0d  
      SAK (SEL_RES): 08  
RATS support: no
Guessing size: seems to be a 1024-byte card
Writing 63 blocks |...............................................................|
Done, 63 of 64 blocks written.
tl@alpine-on-gk41 ~/P/l/utils (master)> sudo ./nfc-mfclassic W a u ~/Downloads/apartment.card
NFC reader: microBuilder.eu opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 22  c7  eb  0d  
      SAK (SEL_RES): 08  
RATS support: no
Guessing size: seems to be a 1024-byte card
Sent bits:     50  00  57  cd  
Sent bits:     40 (7 bits)
Warning: Unlock command [1/2]: failed / not acknowledged.
Trying to rewrite block 0 on a direct write tag.
Writing 64 blocks |................................................................|
Done, 64 of 64 blocks written.
tl@alpine-on-gk41 ~/P/l/utils (master)> 

```

uebian, 2024-09-01:
I'm encountering the same issue. I found that a quick (but ugly) fix to make libnfc 1.8.0 compatible with gen 2 CUID card (direct write card) is to apply the following patch:
```
diff --git a/utils/nfc-mfclassic.c b/utils/nfc-mfclassic.c
index 244af45..a55ec68 100644
--- a/utils/nfc-mfclassic.c
+++ b/utils/nfc-mfclassic.c
@@ -828,7 +828,7 @@ main(int argc, const char *argv[])
       exit(EXIT_FAILURE);
     }
   } else if (atAction == ACTION_WRITE) {
-    if (!write_card(unlock)) {
+    if (!write_card(true)) {
       nfc_close(pnd);
       nfc_exit(context);
       exit(EXIT_FAILURE);
```