nfc-tools/libnfc #567: Acr122 does not read libnfc

Yuris20, 2019-10-12:
this is my mistake can anyone help me?

nfc-list: ERROR: Unable to open NFC device: acr122_usb:002:003
error	libnfc.driver.acr122_pcsc	No ACR122 firmware received, Error: 80100016
nfc-list: ERROR: Unable to open NFC device: acr122_pcsc:ACS ACR122U PICC Interface 00 00

For the moment I have only installed this on Kali Linux 

 `  apt-get install autoconf libtool libusb-dev libpcsclite-dev build-essential

wget https://github.com/nfc-tools/libnfc/releases/download/libnfc-1.7.1/libnfc-1.7.1.tar.bz2

tar -jxvf libnfc-1.7.1.tar.bz2
cd libnfc-1.7.1
autoreconf -vis
./configure --with-drivers=all --sysconfdir=/etc --prefix=/usr
make
sudo make install

sudo nano /etc/modprobe.d/blacklist-libnfc.conf
deve contenere queste righe:
blacklist nfc
blacklist pn533
blacklist pn533_usb   `

kylennn, 2019-10-13:
Hi,

Are you running ACR122 usb or PCSC? 

br

H5117, 2019-12-23:
Some details from my system.

With pcscd stopped:
```
$ nfc-scan-device -v
nfc-scan-device uses libnfc 1.7.1
No NFC device found.
```
With pcscd started, but without a tag on the reader:
```
$ nfc-scan-device -v
nfc-scan-device uses libnfc 1.7.1
1 NFC device(s) found:
error	libnfc.driver.acr122_pcsc	No ACR122 firmware received, Error: 80100016
nfc_open failed for acr122_pcsc:ACS ACR122U 00 00
```
With pcscd started and with a tag on  the reader:
```
$ nfc-scan-device -v
nfc-scan-device uses libnfc 1.7.1
1 NFC device(s) found:
- ACS ACR122U 00 00 / ACR122U214:
    acr122_pcsc:ACS ACR122U 00 00
chip: PN532 v1.6
initator mode modulations: ISO/IEC 14443A (424 kbps, 212 kbps, 106 kbps), FeliCa (424 kbps, 212 kbps), ISO/IEC 14443-4B (106 kbps), ISO/IEC 14443-4B' (106 kbps), ISO/IEC 14443-2B ST SRx (106 kbps), ISO/IEC 14443-2B ASK CTx (106 kbps), Innovision Jewel (106 kbps), Thinfilm NFC Barcode (106 kbps), D.E.P. (424 kbps, 212 kbps, 106 kbps)
target mode modulations: ISO/IEC 14443A (424 kbps, 212 kbps, 106 kbps), FeliCa (424 kbps, 212 kbps), D.E.P. (424 kbps, 212 kbps, 106 kbps)
```
Archlinux
libnfc 1.7.1+204+g4ae4cc8-1
pcsclite 1.8.25-1
acsccid 1.1.7-1
ID 072f:2200 Advanced Card Systems, Ltd ACR122U

neomilium, 2021-01-04:
@H5117 Thanks for your help. No reply from author, closing issue.

closed 2021-01-04 by neomilium