nfc-tools/libnfc #569: Unable to write to USB (Connection timed out) - Virtual Machine - ACR122

kylennn, 2019-10-13:
Hi,

There was a ticket regarding this in 2016 but it has been closed without resolve.
So i hope its okey to open a new one. {old case:https://github.com/nfc-tools/libnfc/issues/369}

Card reader: ACR122 USB
OS: VM Kali Linux 64 bits

Ive blacklisted the readers PN533, PN533_usb and NFC.

Im recieving this error {full log}

`sudo LIBNFC_LOG_LEVEL=3  nfc-list
info	libnfc.config	Unable to open file: /usr/local/etc/nfc/libnfc.conf
debug	libnfc.config	Unable to open directory: /usr/local/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug	libnfc.driver.acr122_usb	device found: Bus 001 Device 030 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:001:030"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00 
error	libnfc.driver.acr122_usb	Unable to write to USB (Connection timed out)
debug	libnfc.general	Unable to open "acr122_usb:001:030".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:001:030
`

Ive read everything i can about this the last week, and it seems like the issue may be that its running in a Virtual Machine. I will try this tomorrow on a USB install with the exact same settings and get back with results.

Hopefully we can solve this together, if you have any ideas dont hesitate to add a comment :)

thekix, 2019-10-14:
Hello,

please, check this: https://github.com/nfc-tools/libnfc/pull/563

Regards,
kix

kylennn, 2019-10-16:
> Hello,
> 
> please, check this: #563
> 
> Regards,
> kix

Hi,

Tried this, still same error.

Just now finished testing with USB booted laptop with the exact same settings as on my VM. And it works. 

I will try the USB boot on my stationary computer as well tommorow. In that case we may be able to confirm its an issue in VM's. 

BR

kylennn, 2019-10-24:
Hi again,

After some intense testing it seems like the issue is the VM.
It works with the exact same settings for USB live, and also for a fresh full install.

Im unsure if this is priority, but maybe add something to the READEME file regarding Virtual Machines. 

BR

AlexeyAV87, 2020-10-08:
Hi frends, i resolve this question !!!!

0) put packages, the seller sent from aliexpress.com package list (I buy ACR122U-A9 from aliexpress.com )
apt-get install pcsc-tools
apt-get install pcscd
apt-get install libacsccid1
apt-get install --reinstall libpcsclite1

1) Unpack the libnfc-1.7.1.tar archive to the root of the disk /

2) Install the libusb-dev package from the distribution repositories

3) We edit the file /libnfc-1.7.1/libnfc/drivers/acr122_usb.c, find and comment on this line, there should be line 430:

/*
     res = usb_set_altinterface(data.pudh, 0);
      if (res < 0) {
        log_put(LOG_GROUP, LOG_CATEGORY, NFC_LOG_PRIORITY_ERROR, "Unable to set alternate setting on USB interface (%s)", _usb_strerror(res));
        usb_close(data.pudh);
        // we failed to use the specified device
        goto free_mem;
      }
*/

4) Excecute
./configure --prefix=/usr --sysconfdir=/etc --with-drivers=acr122_pcsc,acr122_usb,acr122s,arygon

5) Making sure the configuration is correct

Selected drivers:
   acr122_pcsc...... yes
   acr122_usb....... yes
   acr122s.......... yes
   arygon........... yes
   pn53x_usb........ no
   pn532_uart....... no
   pn532_spi.......  no
   pn532_i2c........ no

6) If everything is fine, do
make

7) If everything is fine, do
make install

8) We can specify an explicit path for the location of libnfc.so
export LD_LIBRARY_PATH=/usr/lib

9) Editing file:

/etc/modprobe.d/blacklist-libnfc.conf

should be the following, add / change 

#blacklist nfc
blacklist pn533
blacklist pn533_usb

10) Excecute

modprobe nfc

11) pcscd - should be off
killall -9 pcscd

12) if there is something in the output of lsmod | grep pn
bang, modprobe pn533_usb -r, should be empty lsmod | grep pn

VortixDev, 2021-02-15:
This seems to be caused by the `usb_reset` call that occurs in `acr122_usb_open`; however, it does not appear to be a libnfc bug. I am experiencing this issue myself, having the same profile as OP (Kali VM, ACR122, identical debugging logs).

When the ACR122U is initially connected to the VM, running `pcsc_scan` works fine - it can read the cards. When running `nfc-list`, as part of the `acr122_usb_open` routine, `usb_reset` is called. After this, the ACR122U appears to become unresponsive. Running `pcsc_scan` after this will not work, and the reader needs to be re-plugged to work again.

Compiling and running the USB reset code at https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line confirms that this is an issue with the USB reset, rather than with libnfc. Plugging in the device, then running `pcsc_scan`, followed by the `usbreset` program, followed by another `pcsc_scan` results in the first `pcsc_scan` call working, and the last one not.

I am unsure what is causing the reset to present this problem.

joetomasone, 2021-03-11:
I am having this issue too on a Ubuntu 18.04 KTS VM running under VirtualBox on a Windows 10 host.  

I've tried everything in this thread and cannot get it to work.  It works under the same Ubuntu installation on bare metal.  

[ 3139.227548] usb 1-2: reset full-speed USB device number 3 using ohci-pci
[ 3194.983407] usb 1-2: reset full-speed USB device number 3 using ohci-pci
[ 3200.642213] usb 1-2: USB disconnect, device number 3
[ 3206.731625] usb 1-2: new full-speed USB device number 4 using ohci-pci
[ 3207.233660] usb 1-2: New USB device found, idVendor=072f, idProduct=2200
[ 3207.233667] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3207.233672] usb 1-2: Product: ACR122U PICC Interface
[ 3207.233677] usb 1-2: Manufacturer: ACS
[ 3209.511489] usb 1-2: reset full-speed USB device number 4 using ohci-pci
[ 3410.949186] usb 1-2: reset full-speed USB device number 4 using ohci-pci
root@ubuntu:/home/joe/libnfc# nfc-scan-device
nfc-scan-device uses libnfc libnfc-1.8.0-48-g2b5ad9c
1 NFC device(s) found:
error   libnfc.driver.acr122_usb        Unable to write to USB (Connection timed out)
nfc_open failed for acr122_usb:001:004
root@ubuntu:/home/joe/libnfc#

R2MGaming, 2021-05-18:
I'm having the same issue like joetomasone has with his issue on a Linux virtual machine on Windows 10. I tried typing nfc-list and it gave me the same error below:

nfc-list uses libnfc 1.7.1
error	libnfc.driver.acr122_usb	Unable to write to USB (Connection timed out)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:002:006

Is there a way to fix these on a virtual machine or do I really need to get an expensive linux computer?

tomot21, 2021-11-19:
Hello, same problem here ACR122U + Linux Kali on Vmware Workstation.
Any ideas ?
Thomas

kylennn, 2021-11-19:
Hey,

 Naa sorry, i gave up on this a long time ago.
The LIBNFC seems kinda dead and the ACR122 is very very outdated.

I recommend if you wanna continue hacking get the PN532 from AdaFruit.

BR
Martin

joetomasone, 2021-11-19:
I went whole-hog into Proxmark and the Iceman/RRG build.

https://github.com/RfidResearchGroup/proxmark3

gmgunderground, 2022-01-04:
Same issue with ACR122U + Linux Kali on VMWare
There are no way to make it working