nfc-tools/libnfc #570: Problem with ACR122U-A9: Invalid RDR_to_PC_DataBlock frame

apyredev, 2019-10-24:
Hello!
I have got a brand new ACR122U-A9, and I'm trying to make it work.
I first had the error caused by usb_set_altinterface, and I removed the lines, recompiled the lib, and the error was gone, just as #563 says.

Now, I've got a new error: **Invalid RDR_to_PC_DataBlock frame**

Here is the full output:

`NFC device: ACS / ACR122U PICC Interface opened
error	libnfc.driver.acr122_usb	Invalid RDR_to_PC_DataBlock frame
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
`

I'm running a fresh Debian 10 install.

I did everything I could, looked at all issues related to this, however, I could not find the answer.
Is there going to be an update to NFC-Lib to include the new "generation" of readers? I've encountered a lot of people with the same issue.

Thanks for any answer!

thekix, 2019-10-24:
Hi,

could you try this:

https://gist.github.com/danieloneill/3be43d5374c80d89ea73

I don't know if the patch was included in libnfc.

Regards,
kix
--
Rodolfo García Peñas (kix)
http://www.kix.es/

"I asked him once how to change the key bindings and Dave said 'You use the Change Configuration command. On Unix it is abbreviated as cc.' Dave Conroy and Lawrence Stewart.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

apyredev, 2019-10-25:
It works! I modified the source, recompiled, and installed. I installed mfoc after that, but the problem is that apt install libnfc5 with mfoc as a dependency.
I had to install mfoc, then do **sudo dpkg -r --force-depends libnfc5 and then sudo make install**  to finally make this work. Hope this helps others, I'll reply again if I encounter any more problem. Thanks @thekix !

triple2digital, 2019-11-15:
@apyredev - can you provide step by step and detailed instructions on how you resolved this?

triple2digital, 2019-11-15:
Here is what I am getting:

i@raspberrypi:~ $ sudo nfc-poll 
nfc-poll uses libnfc 1.7.1
NFC reader: ACS / ACR122U PICC Interface opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
error	libnfc.driver.acr122_usb	Invalid RDR_to_PC_DataBlock frame
nfc_initiator_poll_target: Input / Output Error

Rombersoft, 2021-05-28:
> Hi there ! I had the exact same issue, so I applied the fix mentionned in the gist above, recomplied and it worked ! I made a [fork featuring the fix](https://github.com/sam1902/libnfc), if you still have this issue try this and let me know if it fixes it.
> 
> Here are the build instructions I use, make sure the ACR122U is plugged in before starting:
> 
> ```shell
> sudo modprobe -r pn533_usb pn533 nfc
> sudo vim /etc/modprobe.d/blacklist-libnfc.conf
> # (add blacklist lines for nfc, pn533 and pn533_usb)
> sudo apt update
> sudo apt install git binutils make csh g++ sed gawk autoconf automake autotools-dev libglib2.0-dev liblzma-dev libtool
> git clone https://github.com/sam1902/libnfc
> cd libnfc
> sudo autoreconf -is
> sudo ./configure --prefix=/usr --sysconfdir=/etc
> sudo make -j $(nproc)
> sudo make install
> cd /usr/lib
> sudo cp -p libnfc.* x86_64-linux-gnu/
> ```
> 
> Cheers

I have the same problem. I use nfc-frog application (https://github.com/cuamckuu/nfc-frog.git).
On several banking cards on several computers I get this error:
```
error	libnfc.driver.acr122_usb	Unable to read from USB (Operation not permitted)
error	libnfc.driver.acr122_usb	Invalid RDR_to_PC_DataBlock frame
```
Arch Linux OS

Your solution does not help, unfortunately...

closed 2021-09-08 by apyredev

d0ub1l, 2022-01-13:
> Hi there ! I had the exact same issue, so I applied the fix mentionned in the gist above, recomplied and it worked ! I made a [fork featuring the fix](https://github.com/sam1902/libnfc), if you [...]

This one worked for me ! I'm on Ubuntu 20.04 and it's all good. Thanks !

rfmgob, 2022-03-09:
> Hi there ! I had the exact same issue, so I applied the fix mentionned in the gist above, recomplied and it worked ! I made a [fork featuring the fix](https://github.com/sam1902/libnfc), if you [...]

Finally something that fixes my issues with this reader. Thanks a million!

sachcnt, 2024-02-05:
> Finally something that fixes my issues with this reader. Thanks a million!

Try it on kali 2023.4 and still same issue. All steps are well installed but at the end it's not working. Any clue?

Rombersoft, 2024-02-06:
My problem went away when I refused libnfc and began to use PCSC.dll from Nuget (C# project)