nfc-tools/libnfc #574: ACR1252U-M1, nfc-list: NFC device not found.

zhaxylykbayev, 2019-11-08:
I know its not an issue, but need help. 
Firstly, does libnfc compatible with ACR1252U ?
I'm using version 1.7.1 and tried on Mac OS 10.15, fully fresh ubuntu 18.04 but no luck.
Reader Im using- https://www.acs.com.hk/en/products/342/
I have tried everything on stackoverflow and github with no results

thekix, 2019-11-08:
Hi,

please, check it with this patch: https://github.com/nfc-tools/libnfc/pull/563

Remember to clean the previous libnfc packages to avoid wrong library selection.

If the problem is solved with this patch, please, merge the issues.

Regards,
kix

zhaxylykbayev, 2019-11-11:
Deleting usb_set_altinterface didn't help, even on clean ubuntu
steps i followed:
1) install fresh ubuntu
2) clone last version 1.7.1
3) go to library folder > libnfc > drivers> acr122_usb.c deleted
```  res = usb_set_altinterface(data.pudh, 0);
      if (res < 0) {
        log_put(LOG_GROUP, LOG_CATEGORY, NFC_LOG_PRIORITY_ERROR, "Unable to set alternate setting on USB interface (%s)", _usb_strerror(res));
        usb_close(data.pudh);
        // we failed to use the specified device
        goto free_mem;
      }
```

4) install all required libraries
5) and in library folder typed these commands:
```
autoreconf -vis
./configure --with-drivers=acr122_usb --prefix=/usr
make
sudo make install
```

thekix, 2019-11-11:
Did you remove this line too?:

`      res = usb_set_altinterface(data.pudh, 0);`

You need remove these lines:

```
      res = usb_set_altinterface(data.pudh, 0);
      if (res < 0) {
        log_put(LOG_GROUP, LOG_CATEGORY, NFC_LOG_PRIORITY_ERROR, "Unable to set alternate setting on USB interface (%s)", _usb_strerror(res));
        usb_close(data.pudh);
        // we failed to use the specified device
        goto free_mem;
      }

```

zhaxylykbayev, 2019-11-11:
Yes, sorry, i've mistaken here. I deleted all these lines including 
```
res = usb_set_altinterface(data.pudh, 0);
```

zhaxylykbayev, 2019-11-13:
Closing issue, didn't find any solution with acr reader. Just bought pn532 and it's working

closed 2019-11-13 by zhaxylykbayev