nfc-tools/libnfc #576: ACR122U-A9 "Unable to open "acr122_usb:001:003"

oreineke, 2019-12-13:
Hello!

I have a new ACR122U-A9 reader (S/N: RR171-399...) that I can not get working with libnfc:
```
# uname -a
Linux kali 5.3.0-kali2-amd64 #1 SMP Debian 5.3.9-3kali1 (2019-11-20) x86_64 GNU/Linux

# LIBNFC_LOG_LEVEL=3 nfc-list
debug   libnfc.config   Unable to open directory: /etc/nfc/devices.d
debug   libnfc.general  log_level is set to 3
debug   libnfc.general  allow_autoscan is set to true
debug   libnfc.general  allow_intrusive_scan is set to false
debug   libnfc.general  0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug   libnfc.driver.acr122_usb        device found: Bus 001 Device 037 Name ACS ACR122
debug   libnfc.general  1 device(s) found using acr122_usb driver
debug   libnfc.general  0 device(s) found using pn53x_usb driver
debug   libnfc.driver.acr122_usb        3 element(s) have been decoded from "acr122_usb:001:037"
debug   libnfc.driver.acr122_usb        TX: 62 00 00 00 00 00 00 01 00 00
debug   libnfc.driver.acr122_usb        RX: 80 02 00 00 00 00 00 02 81 00 3b 00
debug   libnfc.driver.acr122_usb        ACR122 PICC Operating Parameters
debug   libnfc.driver.acr122_usb        TX: 6f 05 00 00 00 00 00 00 00 00 ff 00 51 00 00
debug   libnfc.general  Unable to open "acr122_usb:001:037".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:001:037
```
I am aware that others have reported issues with these newer devices and have tried all possible solutions I could find here:

I have unloaded and blacklisted the usual modules: `ncf`, `pn533` and `pn533_usb`. I have tired with service `pcscd`running and stopped.

I have tried `ifdDriverOptions` `0x0001`, `0x0004`, and `0x0005` in `/etc/libccid_Info.plist`.

I have built the lib from `https://github.com/thekix/libnfc` with PR#563. I also tried patch https://gist.github.com/danieloneill/3be43d5374c80d89ea73.

Nothing helped.

With `pcsc_scan`, I can see the reader just fine:
```
# pcsc_scan
PC/SC device scanner
V 1.5.2 (c) 2001-2017, Ludovic Rousseau <ludovic.rousseau@free.fr>
Using reader plug'n play mechanism
Scanning present readers...
0: ACS ACR122U PICC Interface (A02014090501) 00 00
1: ACS ACR122U PICC Interface (A02014090501) 00 01
2: Alcor Micro AU9560 01 00

Wed Dec 11 10:07:36 2019
 Reader 0: ACS ACR122U PICC Interface (A02014090501) 00 00
  Card state: Card removed,
 Reader 1: ACS ACR122U PICC Interface (A02014090501) 00 01
  Card state: Card removed,
 Reader 2: Alcor Micro AU9560 01 00
  Card state: Card removed,

Wed Dec 11 10:08:09 2019
 Reader 0: ACS ACR122U PICC Interface (A02014090501) 00 00
  Card state: Card inserted,
  ATR: 3B 8F 80 01 80 4F 0C A0 00 00 03 06 03 00 01 00 00 00 00 6A

ATR: 3B 8F 80 01 80 4F 0C A0 00 00 03 06 03 00 01 00 00 00 00 6A
+ TS = 3B --> Direct Convention
+ T0 = 8F, Y(1): 1000, K: 15 (historical bytes)
  TD(1) = 80 --> Y(i+1) = 1000, Protocol T = 0 
-----
  TD(2) = 01 --> Y(i+1) = 0000, Protocol T = 1 
-----
+ Historical bytes: 80 4F 0C A0 00 00 03 06 03 00 01 00 00 00 00
  Category indicator byte: 80 (compact TLV data object)
    Tag: 4, len: F (initial access data)
      Initial access data: 0C A0 00 00 03 06 03 00 01 00 00 00 00
+ TCK = 6A (correct checksum)

Possibly identified card (using /usr/share/pcsc/smartcard_list.txt):
3B 8F 80 01 80 4F 0C A0 00 00 03 06 03 00 01 00 00 00 00 6A
3B 8F 80 01 80 4F 0C A0 00 00 03 06 .. 00 01 00 00 00 00 ..
        Mifare Standard 1K (as per PCSC std part3)
3B 8F 80 01 80 4F 0C A0 00 00 03 06 03 00 01 00 00 00 00 6A
3B 8F 80 01 80 4F 0C A0 00 00 03 06 03 .. .. 00 00 00 00 ..
        RFID - ISO 14443 Type A Part 3 (as per PCSC std part3)
3B 8F 80 01 80 4F 0C A0 00 00 03 06 03 00 01 00 00 00 00 6A
        Philips MIFARE Standard (1 Kbytes EEPROM)
        http://www.nxp.com/#/pip/pip=[pfp=41863]|pp=[t=pfp,i=41863]
        RFID - ISO 14443 Type A - Transport for London Oyster
        ACOS5/1k Mirfare
        RFID - ISO 14443 Type A - NXP Mifare card with 1k EEPROM
        vivotech ViVOcard Contactless Test Card
        Bangkok BTS Sky SmartPass
        Mifare Classic 1K (block 0 re-writeable)
 /  - 
```

I also tried Ubuntu 18.04.

If anybody has any ideas what else I could try to get the reader working that would great!

geeto, 2020-01-17:
I am having the exact same problem. Any luck?

oreineke, 2020-01-17:
I came one step closer to getting it working by re-building libnfc after a 

`/configure --with-drivers=arc122_pcsc`

However, my reader still did not work correctly as it turned out to be a fake Chinese clone. With an original ACR122U-A9 though, it worked fine then (even in a virtual machine).

NikoJako, 2020-01-18:
I had a problem as well where my ACR122 reader wouldn’t work-meaning it
wouldn’t scan any NFC card I applied to the reader. I forgot the error
message libnfc was giving but it was pretty strait forward.

The solution was to open the software that ACR provided with the reader and
enable the reader antenna!! Who would have thought The reader antenna was
disabled by default. If your ACR didn’t come with a CD I’m sure you can
download from the ACR website

oreineke, 2020-01-18:
> I had a problem as well where my ACR122 reader wouldn’t work-meaning it
wouldn’t scan any NFC card I applied to the reader. 

Interesting, but a completely different problem and therefore should not be posted in this issue.

GillesJ, 2020-01-25:
Same issue here on Ubuntu 18.04. Tried all the work-arounds listed here. pcsc_scan works fine too.

oreineke, 2020-01-27:
Are you using a Chinese fake ACR122, e.g. from eBay? With such a fake, I also had no luck getting it to work with libnfc.

If, however, you have an original device, did you re-build libnfc after a

`./configure --with-drivers=acr122_pcsc`

and then set `LD_LIBRARY_PATH` to point to the freshly build libnfc?

GillesJ, 2020-01-27:
> Are you using a Chinese fake ACR122, e.g. from eBay? With such a fake, I also had no luck getting it to work with libnfc.

I bought it from Amazon.de from a reputable seller, but I will double check if the reader is genuine by contacting ACS with the serial number. Do you know of any way to check if an ACR122 is a counterfeit besides contacting ACS? All ACS has on the topic of counterfeit readers is [this statement](https://www.acs.com.hk/en/press-release/2266/advanced-card-systems-ltd.-declaration-against-counterfeit-acr122u-nfc-readers/), but no online validation tool of sorts. 

> If, however, you have an original device, did you re-build libnfc after a
> 
> `./configure --with-drivers=acr122_pcsc`

I did this, alongside all fixes described in OP.

> and then set `LD_LIBRARY_PATH` to point to the freshly build libnfc?

But I did not do this and will test this in a few hours. Will get back to you.

oreineke, 2020-01-27:
Don't get you hopes up regarding an answer from ACS: I asked them about my reader and they never replied.
In my case, the fake Chinese reader had an ACS logo on the back(!) side that had the "a" in the wrong front. Hard to tell though, if you can not compare with an original.
The `LD_LIBRARY_PATH` is just to make sure the libnfc you build is actually used.

AlexeyAV87, 2020-10-09:
Hi frends, i resolve this question !!!!

0) put packages, the seller sent from aliexpress.com package list (I buy ACR122U-A9 from aliexpress.com )
apt-get install pcsc-tools
apt-get install pcscd
apt-get install libacsccid1
apt-get install --reinstall libpcsclite1

1) Download official site binary source, Unpack the libnfc-1.7.1.tar archive to the root of the disk /

2) Install the libusb-dev package from the distribution repositories

3) We edit the file /libnfc-1.7.1/libnfc/drivers/acr122_usb.c, find and comment on this line, there should be line 430:

/*
     res = usb_set_altinterface(data.pudh, 0);
      if (res < 0) {
        log_put(LOG_GROUP, LOG_CATEGORY, NFC_LOG_PRIORITY_ERROR, "Unable to set alternate setting on USB interface (%s)", _usb_strerror(res));
        usb_close(data.pudh);
        // we failed to use the specified device
        goto free_mem;
      }
*/

4) Excecute
./configure --prefix=/usr --sysconfdir=/etc --with-drivers=acr122_pcsc,acr122_usb,acr122s,arygon

5) Making sure the configuration is correct

Selected drivers:
   acr122_pcsc...... yes
   acr122_usb....... yes
   acr122s.......... yes
   arygon........... yes
   pn53x_usb........ no
   pn532_uart....... no
   pn532_spi.......  no
   pn532_i2c........ no

6) If everything is fine, do
make

7) If everything is fine, do
make install

8) We can specify an explicit path for the location of libnfc.so
export LD_LIBRARY_PATH=/usr/lib

9) Editing file:

/etc/modprobe.d/blacklist-libnfc.conf

should be the following, add / change 

#blacklist nfc
blacklist pn533
blacklist pn533_usb

10) Excecute

modprobe nfc

11) pcscd - should be off
killall -9 pcscd

12) if there is something in the output of lsmod | grep pn
bang, modprobe pn533_usb -r, should be empty lsmod | grep pn

closed 2021-01-04 by smortex