nfc-tools/libnfc #582: Setting 7 byte UID of Mifare Classic

asemchenko, 2020-01-24:
Hi! 

Is there any utility or just a piece of code that allows set 7 byte UID of Mifare magic card?

 I have founded [`nfc-mfsetuid`](https://github.com/nfc-tools/libnfc/blob/master/examples/nfc-mfsetuid.c) util in this library but it works only with 4 byte UID.

gebi, 2020-09-04:
hi, i just hit the same issue and then calculated the sum "by hand" and wrote the whole block.

eg. https://www.nxp.com/docs/en/application-note/AN10927.pdf

doegox, 2020-09-04:
Can you elaborate?
You used the "3.2.2 Double Size UID for card activation, but Single Size NUID for the system" and wrote the 4b UID on a 4b magic card?
Or you're talking about a 7b magic card? In which case what did you do exactly?

gebi, 2020-09-07:
@doegox i honestly don't know if i have 4 or 7 byte cards.

`nfc-list` only shows 4 bytes, but as all other tools also only seem to support 4 byte UIDs i'm not sure if it's just not displaying the rest.

sorry for the confusion, i only needed to clone the UID and instead of doing something with 4 or 7 byte UID's i just wrote the whole block0 (which `nfc-mfsetuid` also supports and which works, though there it ends :/.
All other functions eg. nfc-mfclassic seems to be broken because no key's are ever written to the card but no errors are shown. If i read back the card all keys (a/b) are still 0xff. The cards itself seem to work because the android mifare classic app can write/clone evertying except block0 just fine (the are magic cards v1 it seems).
And i'm not sure on how to debug this or how to help here to get that working again.

doegox, 2020-09-07:
ok thanks for the feedback anyway @gebi

xavave, 2023-01-22:
Hi,
Im' trying to add 7 bytes support on nfc-mfsetuid: my commit is here:
https://github.com/xavave/libnfc_with_extra_tools/commit/26733fd23110dc153c496788f2068f79f953358e#diff-b2c594415cfd2530164c9cf2c2f096c2d1d8fc62a09772c264de3ff9ee653117

I used this for help: https://stackoverflow.com/questions/37837730/mifare-cards-distinguish-between-4-byte-and-7-byte-uids
and also this: https://www.nxp.com/docs/en/application-note/AN10927.pdf
however, I don't have a 7 bytes UID tag for testing
I've tried to use my forked nfc-mfsetuid on a 4 bytes tag, but it just bricked it (it's not recognized anymore then)
I should try to detect if the tag is 4 bytes or 7 bytes UID before trying to write
[image]
Should I try to convert the 7 bytes UID to 4 bytes FNUID?
The conversion results in a weird UID:
[image]
If someone as some ideas to solve it, please comment

gmkbenjamin, 2024-08-02:
> Hi, Im' trying to add 7 bytes support on nfc-mfsetuid: my commit is here: [xavave@26733fd#diff-b2c594415cfd2530164c9cf2c2f096c2d1d8fc62a09772c264de3ff9ee653117](https://github.com/xavave/libnfc_with_extra_tools/commit/26733fd23110dc153c496788f2068f79f953358e#diff-b2c594415cfd2530164c9cf2c2f096c2d1d8fc62a09772c264de3ff9ee653117)
> 
> I used this for help: https://stackoverflow.com/questions/37837730/mifare-cards-distinguish-between-4-byte-and-7-byte-uids and also this: https://www.nxp.com/docs/en/application-note/AN10927.pdf however, I don't have a 7 bytes UID tag for testing I've tried to use my forked nfc-mfsetuid on a 4 bytes tag, but it just bricked it (it's not recognized anymore then) I should try to detect if the tag is 4 bytes or 7 bytes UID before trying to write should I recalculate the CRC for 7 bytes UID? the method may be different than for 4-bytes abtData[4] = abtData[0] ^ abtData[1] ^ abtData[2] ^ abtData[3]; [image] Should I try to convert the 7 bytes UID to 4 bytes FNUID? The conversion results in a weird UID: [image] If someone as some ideas to solve it, please comment

Pump this. Tried writing 7 byte uid to a tag and soft bricked it.