nfc-tools/libnfc #584: ACR 122u on macOS Catalina

alerighi, 2020-02-16:
Hi, I've tried using libnfc on macOS Catalina unsuccessfully . 

I've followed the indication in the issue #330 and I now get at this point:
```
nfc-list uses libnfc 1.7.1
error	libnfc.driver.acr122_usb	Unable to set alternate setting on USB interface (Result too large)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:020:012
```

I have libnfc installed with HomeBrew, I've tried compiling it from sources and it fails both with autotools and with CMake.

simone-compagnone, 2020-02-21:
Hi,

try this:
1. sudo launchctl remove com.apple.ifdreader
2. sudo launchctl stop com.apple.ifdreader
3. plugin your device 
4. sudo nfc-list

neomilium, 2021-01-04:
@simone-compagnone Thanks for your help. No reply from author, closing the issue.

closed 2021-01-04 by neomilium