nfc-tools/libnfc #589: nfc-list: ERROR: Unable to open NFC device: acr122_usb:002:008

anonsurfjo, 2020-04-02:
hello ! all
i can't open NFC reader on Kali Linux 2020 
i have this Error :NFC-list: ERROR: Unable to open NFC device: acr122_USB:002:008
what is problem ??

kino-dome, 2020-04-30:
Same here. It hangs unless I eject the usb and then the error shows up.

Makcell96, 2021-01-31:
Salut a tous je rencontre aussi le meme probleme apres l'installation de mon lecteru/ecrivain sur xubuntu.
il m'apparait le message d'eurreur si dessous quand je fait la commande sudo nfc-list.je vous remerci de bien vouloir m'aider svp:
error libnfc.driver.acr122_usb    Unable to set alternate setting on USB interface (Connection timed out) 
nfc-list: ERROR: Unable to open NFC device: acr122_usb:008:002