nfc-tools/libnfc #593: INSTALL file not present, cannot install

Nekkowe, 2020-04-14:
In the Readme, the install instructions are delegated to a file that is not in the repo, but listed in .gitignore instead:
`See the file INSTALL for configure, build and install details.`
Also in the .gitignore are `configure` and `Makefile`.
As it stands, I have no idea how to build and install libnfc from source.

Nekkowe, 2020-04-15:
Since there is no `configure` or makefile present, the build instructions of

> It should be as simple as running these two commands:
>./configure
>make

also do not work.

LudovicRousseau, 2020-05-22:
You can use the `autoreconf` command to create the `configure` script.
The released `.tar.bz2` archive already contains the `configure` script.

Or you can use `cmake .` if you prefer to use cmake.

abathur8bit, 2020-06-24:
Running on a Raspberry PI with Rasbain, fresh clone, autoreconf generates a the following errors. It generated a configure script, but trying to run it produces 

```
configure: error: cannot find install-sh, install.sh, or shtool in "." "./.." "./../.."
```

Running autoreconf:

```
pi@mikey:~/workspace/nfc/libnfc$ autoreconf
configure.ac:17: error: required file './ar-lib' not found
configure.ac:17:   'automake --add-missing' can install 'ar-lib'
configure.ac:17: error: required file './compile' not found
configure.ac:17:   'automake --add-missing' can install 'compile'
configure.ac:16: error: required file './install-sh' not found
configure.ac:16:   'automake --add-missing' can install 'install-sh'
configure.ac:16: error: required file './missing' not found
configure.ac:16:   'automake --add-missing' can install 'missing'
examples/Makefile.am: error: required file './depcomp' not found
examples/Makefile.am:   'automake --add-missing' can install 'depcomp'
libnfc/Makefile.am:6: error: Libtool library used but 'LIBTOOL' is undefined
libnfc/Makefile.am:6:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libnfc/Makefile.am:6:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libnfc/Makefile.am:6:   If 'LT_INIT' is in 'configure.ac', make sure
libnfc/Makefile.am:6:   its definition is in aclocal's search path.
libnfc/buses/Makefile.am:4: error: Libtool library used but 'LIBTOOL' is undefined
libnfc/buses/Makefile.am:4:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libnfc/buses/Makefile.am:4:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libnfc/buses/Makefile.am:4:   If 'LT_INIT' is in 'configure.ac', make sure
libnfc/buses/Makefile.am:4:   its definition is in aclocal's search path.
libnfc/chips/Makefile.am:5: error: Libtool library used but 'LIBTOOL' is undefined
libnfc/chips/Makefile.am:5:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libnfc/chips/Makefile.am:5:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libnfc/chips/Makefile.am:5:   If 'LT_INIT' is in 'configure.ac', make sure
libnfc/chips/Makefile.am:5:   its definition is in aclocal's search path.
libnfc/drivers/Makefile.am:4: error: Libtool library used but 'LIBTOOL' is undefined
libnfc/drivers/Makefile.am:4:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libnfc/drivers/Makefile.am:4:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libnfc/drivers/Makefile.am:4:   If 'LT_INIT' is in 'configure.ac', make sure
libnfc/drivers/Makefile.am:4:   its definition is in aclocal's search path.
test/Makefile.am:21: error: Libtool library used but 'LIBTOOL' is undefined
test/Makefile.am:21:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
test/Makefile.am:21:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
test/Makefile.am:21:   If 'LT_INIT' is in 'configure.ac', make sure
test/Makefile.am:21:   its definition is in aclocal's search path.
parallel-tests: error: required file './test-driver' not found
parallel-tests:   'automake --add-missing' can install 'test-driver'
utils/Makefile.am:15: error: Libtool library used but 'LIBTOOL' is undefined
utils/Makefile.am:15:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
utils/Makefile.am:15:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
utils/Makefile.am:15:   If 'LT_INIT' is in 'configure.ac', make sure
utils/Makefile.am:15:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1

```

LudovicRousseau, 2020-06-24:
You should use https://github.com/nfc-tools/libnfc/releases/download/libnfc-1.8.0/libnfc-1.8.0.tar.bz2 available from https://github.com/nfc-tools/libnfc/releases

This archive already contains the `configure` script.

closed 2021-01-04 by smortex

davet2001, 2021-03-30:
Please can the github README.md be updated, to say something more helpful?

I agree that autoreconf is a valid way of generating the configure script.  But README.md needs to at least give people a clue, and when people look at github they are not looking at the content of the tar.gz release.

E.g. directly under the Installation title:

Note: If working directly from a git clone of the repository, some of the files need to be generated first. Run 
autoreconf <parameters, etc>
or use 
<link to tar.gz download>

LudovicRousseau, 2021-03-31:
@davet2001 please propose a pull request.

tombolinux, 2021-06-07:
> You can use the `autoreconf` command to create the `configure` script.
> The released `.tar.bz2` archive already contains the `configure` script.
> 
> Or you can use `cmake .` if you prefer to use cmake.

for my hardware: pn532
After untar the archive:
`cd libnfc-libnfc-1.8.0
mkdir /etc/nfc
mkdir /etc/nfc/devices.d
cp contrib/libnfc/pn532_uart_on_rpi.conf.sample /etc/nfc/devices.d/pn532_uart_on_rpi.conf
autoreconf -i 
vi /etc/nfc/devices.d/pn532_uart_on_rpi.conf`
add the following line
`allow_intrusive_scan = true`
then
`./configure --sysconfdir=/etc --prefix=/usr
make
make install all`