nfc-tools/libnfc #596: Please cut a new release

fuzxxl, 2020-05-03:
It would be great to have a new release some time so open source software can use new libnfc features using a stable library version as shipped by distributors.  The last release is five years old and lacks many of the new features available since.

doegox, 2020-05-21:
done :)

fuzxxl, 2020-05-21:
Oh yeah!

fuzxxl, 2020-05-21:
Though really, given that you changed a number of interfaces in a backward-incompatible manner, the new version should really be 2.0.0 or at least 1.8.0 to indicate the backwards-incompatible changes.  Making such changes in a minor release is not quite in the spirit of the usual versioning conventions.

doegox, 2020-05-21:
Hmm ok yes a lot of changes in 5 years.
@smortex @neomilium @LudovicRousseau before cutting a 1.8.0 is there anything I forgot ? I'm a bit rusted on making releases...
Lib should also become `libnfc.so.6`  I guess.

doegox, 2020-05-21:
libnfc/CMakeLists.txt :
`SET_TARGET_PROPERTIES(nfc PROPERTIES SOVERSION 6 VERSION 6.0.0)`
libnfc/Makefile.am :
`libnfc_la_LDFLAGS = -no-undefined -version-info 6:0:0 ...`
and documenting it in https://github.com/nfc-tools/libnfc/wiki/ReleasingProcess

fuzxxl, 2020-05-21:
Hm... I think there's really no point now that the release is out; it would only confuse people.  Maybe call the next one 1.8.1 then?

fuzxxl, 2020-05-22:
Oh, and btw: by inserting `NMT_BARCODE` into the middle of the `nfc_modulation_type` you made a backwards-incompatible change because the values of all subsequent enumeration constants are shifted.  You should clearly note that somewhere in the release notes so distributions don't ship 1.7.2 as a minor upgrade, breaking all applications compiled against 1.7.1 in the process.

Please avoid this kind of change in the future.

LudovicRousseau, 2020-05-22:
I pushed some minor fixes.
The ABI change is unfortunate. A version 1.8.0 (with new .so version) should be released really soon to avoid upgrade issues.

doegox, 2020-05-22:
done.
https://github.com/nfc-tools/libnfc/releases/tag/libnfc-1.8.0

closed 2020-05-22 by doegox

fuzxxl, 2020-05-22:
Thanks a lot.  Will fix my wrapper code to restore the correct argument order.

AdamLaurie, 2020-05-22:
Yeah, that caught me out too for RFIDIot!

> Thanks a lot. Will fix my wrapper code to restore the correct enumeration order.