nfc-tools/libnfc #602: nfc-barcode failed

KyryloAntoshyn, 2020-05-27:
Hi, when I am trying to build the 1.8.0 version of the library I have the following error: 'nfc-barcode' failed. How to fix it?

[image]

doegox, 2020-05-27:
please provide more information so we can reproduce your issue.

KyryloAntoshyn, 2020-05-27:
I downloaded the source archive, unzipped and issued following commands:
1) sudo autoreconf -vis
2) sudo ./configure --prefix=/usr --sysconfdir=/etc --with-drivers=all
3) sudo make
4) sudo make install

3rd and 4th commands lead to error with libnfc.so no such file or directory.
I tried to do same steps with libnfc 1.7.1 and successfully installed it. However, we need the newest version of libnfc.

doegox, 2020-05-27:
autoreconf -vis   shouldn't be run if you're using https://github.com/nfc-tools/libnfc/releases/download/libnfc-1.8.0/libnfc-1.8.0.tar.bz2
Here on a Debian :
```
wget https://github.com/nfc-tools/libnfc/releases/download/libnfc-1.8.0/libnfc-1.8.0.tar.bz2
tar xf libnfc-1.8.0.tar.bz2 
cd libnfc-1.8.0/
./configure --prefix=/usr --sysconfdir=/etc --with-drivers=all
make
sudo make install
```
works smoothly.

If this still fails, please provide more information on the platform, OS, OS version.

KyryloAntoshyn, 2020-05-27:
Thanks, now it works on my machine.

closed 2020-05-27 by KyryloAntoshyn