nfc-tools/libnfc #604: Unable to compile libnfc sample in 1.8.0

KyryloAntoshyn, 2020-05-28:
Hi, I copied [sample](http://www.libnfc.org/api/examples_page.html) into the file and try to compile it using following command:
gcc -o libnfc_example libnfc_example.c -lnfc

However, compiler says about undefined references inside libnfc. How to fix it?

I am using Linux raspberrypi 4.19.66-v7+.

[image]

Thanks in advance.

KyryloAntoshyn, 2020-05-28:
It is fixed, it works by linking:  gcc -o libnfc_example libnfc_example.c -lnfc -lusb.

However, I faced another problem with this sample. If you run it and card is not on the reader it says following error: "error	libnfc.driver.acr122_usb	Unable to write to USB (Connection timed out)."
As I remember before new version if the tag was not on the card it was waiting it without error messages.

[image]

How to fix it?

KyryloAntoshyn, 2020-05-28:
Try my sample, please
```
// To compile this simple example:
// $ gcc -o quick_start_example1 quick_start_example1.c -lnfc

#include <stdlib.h>
#include <nfc/nfc.h>
//#include <freefare.h>

static void
print_hex(const uint8_t *pbtData, const size_t szBytes)
{
  size_t  szPos;

  for (szPos = 0; szPos < szBytes; szPos++) {
    printf("%02x  ", pbtData[szPos]);
  }
  printf("\n");
}

int
main(int argc, const char *argv[])
{
  nfc_device *pnd;
  nfc_target nt;

  // Allocate only a pointer to nfc_context
  nfc_context *context;

  // Initialize libnfc and set the nfc_context
  nfc_init(&context);
  if (context == NULL) {
    printf("Unable to init libnfc (malloc)\n");
    exit(EXIT_FAILURE);
  }

  // Display libnfc version
  const char *acLibnfcVersion = nfc_version();
  (void)argc;
  printf("%s uses libnfc %s\n", argv[0], acLibnfcVersion);

  // Open, using the first available NFC device which can be in order of selection:
  //   - default device specified using environment variable or
  //   - first specified device in libnfc.conf (/etc/nfc) or
  //   - first specified device in device-configuration directory (/etc/nfc/devices.d) or
  //   - first auto-detected (if feature is not disabled in libnfc.conf) device
  pnd = nfc_open(context, NULL);

  if (pnd == NULL) {
    printf("ERROR: %s\n", "Unable to open NFC device.");
    exit(EXIT_FAILURE);
  }
  // Set opened NFC device to initiator mode
  if (nfc_initiator_init(pnd) < 0) {
    nfc_perror(pnd, "nfc_initiator_init");
    exit(EXIT_FAILURE);
  }

  printf("NFC reader: %s opened\n", nfc_device_get_name(pnd));

  // Poll for a ISO14443A (MIFARE) tag
  const nfc_modulation nmMifare = {
    .nmt = NMT_ISO14443A,
    .nbr = NBR_106,
  };
  
  while(true)
  {
	  if (nfc_initiator_select_passive_target(pnd, nmMifare, NULL, 0, &nt) > 0) {
		printf("The following (NFC) ISO14443A tag was found:\n");
		printf("    ATQA (SENS_RES): ");
		print_hex(nt.nti.nai.abtAtqa, 2);
		printf("       UID (NFCID%c): ", (nt.nti.nai.abtUid[0] == 0x08 ? '3' : '1'));
		print_hex(nt.nti.nai.abtUid, nt.nti.nai.szUidLen);
		printf("      SAK (SEL_RES): ");
		print_hex(&nt.nti.nai.btSak, 1);
		if (nt.nti.nai.szAtsLen) {
		  printf("          ATS (ATR): ");
		  print_hex(nt.nti.nai.abtAts, nt.nti.nai.szAtsLen);
		}
		
		printf("Waiting for card removing...");
		while (0 == nfc_initiator_target_is_present(pnd, &nt)) {}
		nfc_perror(pnd, "nfc_initiator_target_is_present");
		printf("Card was removed");
		
		// // Get freefare tag
		// auto freefareTag = freefare_tag_new(pnd, nt);
		// printf("Freefare tag is: %s", freefareTag);
	  }
  }
  
  // Close NFC device
  nfc_close(pnd);
  // Release the context
  nfc_exit(context);
  exit(EXIT_SUCCESS);
}
```

doegox, 2020-05-28:
Hmm this is strange you had to explicitly link with `-lusb`.
Here :
```
/tmp/libnfc/libnfc-1.8.0/examples/doc$ gcc -o quick_start_example1 quick_start_example1.c -lnfc
/tmp/libnfc/libnfc-1.8.0/examples/doc$ ldd quick_start_example1
	linux-vdso.so.1 (0x00007ffd8cc9c000)
	libnfc.so.6 => /usr/lib/x86_64-linux-gnu/libnfc.so.6 (0x00007fd2e4afe000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd2e493b000)
	libusb-0.1.so.4 => /lib/x86_64-linux-gnu/libusb-0.1.so.4 (0x00007fd2e4730000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fd2e4b74000)
```
Regarding the other error, is it specific to the example code or do you get the same issue with nfc-list, nfc-anticoll,... ?

KyryloAntoshyn, 2020-05-28:
I have this errors only with my sample when calling nfc_initiator_select_passive_target()

closed 2020-05-30 by KyryloAntoshyn