nfc-tools/libnfc #615: Missing shared libraries after install

xLinkOut, 2020-07-25:
After installing `libnfc` by downloading it from the [release page](https://github.com/nfc-tools/libnfc/releases/tag/libnfc-1.8.0),
none of the installed software works, showing the following error:

> nfc-poll: error while loading shared libraries: libnfc.so.6: cannot open shared object file: No such file or directory

This seems to fix the problem [source](https://oneguyoneblog.com/2015/09/16/acr122u-nfc-usb-reader-raspberry-pi/) :
```bash
sudo sh -c "echo /usr/local/lib > /etc/ld.so.conf.d/usr-local-lib.conf"
sudo ldconfig
```
This happens even with 1.7.0, 1.7.1 and 1.8.0.

romain145, 2020-10-15:
Issue and fix confirmed with https://github.com/nfc-tools/libnfc/commit/c3f739dea339a71c59d7d53ab6b0ecc477c3ab73

closed 2021-01-04 by smortex

stanx89, 2021-07-05:
> After installing `libnfc` by downloading it from the [release page](https://github.com/nfc-tools/libnfc/releases/tag/libnfc-1.8.0),
> none of the installed software works, showing the following error:
> 
> > nfc-poll: error while loading shared libraries: libnfc.so.6: cannot open shared object file: No such file or directory
> 
> This seems to fix the problem [source](https://oneguyoneblog.com/2015/09/16/acr122u-nfc-usb-reader-raspberry-pi/) :
> 
> ```shell
> sudo sh -c "echo /usr/local/lib > /etc/ld.so.conf.d/usr-local-lib.conf"
> sudo ldconfig
> ```
> 
> This happens even with 1.7.0, 1.7.1 and 1.8.0.

Was stuck with similar thing, this helped..