nfc-tools/libnfc #617: Setting a longer timeout for detecting tag?

maxieds, 2020-07-30:
I am working with Mifare tags and am getting stuck in the REQA loop when trying to detect the tag with ``nfc-scan-device`` and the ``nfc-tools`` utility ``mifare-desfire-info``. What I think is happening is that for one reason or another (specific to the tag implementation) it is taking too long for ``libnfc`` to get the response back from the REQA (``0x26``) command that it is sending out. The utilities keep issuing a new REQA request to the tag, and it never processes the response from my tag. 

Is there a way to prevent ``libnfc`` from re-issuing the REQA request to the tag? E.g., can I set a very high threshold for the timeout before ``libnfc`` tries to poll again when it thinks there is no response coming back to it? I'm using a reader that is working off the ``pn53_usb`` driver. 

Thanks.

doegox, 2020-07-31:
Anticollision is handled directly by the pn53x firmware.
REQA must receive the reply at a very strict timing according to the ISO.
PN53x are moderately tolerant to slower answers, we can see it when one tries to mount a relay attack against a pn53x. It works so-so...
You can try ``nfc-anticol`` which will operate a pseudo-anticollision purely from libnfc, this one is more lenient to slow answers.
But honestly, besides dumb relay attacks (relaying also the anticollision), I see no reason for a legit tag to answer slowly to a REQA...

maxieds, 2020-07-31:
@doegox 
The reason (I think) that my tag implementation is so slow to respond is that I have **a lot** of slow logging going on for development. It's not really possible to remove all of this right now. 

What I'm really after is being able to write some custom C programs with ``libnfc`` and/or ``libfreefare`` that bypass all of the anticollision and just start directly communicating with the card. Is there a way to do this? I have seen the source to the ``libfreefare`` utilities, and it seems like to get a handle on a tag, you have to go through the problematic anticollison, where I'm running into problems. Is there a good way to just log the raw responses (without driver framing) that are received from the tag?

doegox, 2020-07-31:
Yes, see the nfc-anticol I mentioned: https://github.com/nfc-tools/libnfc/blob/master/examples/nfc-anticol.c

maxieds, 2020-07-31:
@doegox 
Can you help me figure out some issues compiling the ``nfc-anticol.c`` sources locally instead of in my clone of ``libnfc`` and running ``make install``? 

Here is the output of when I run it from the systems binaries:
```bash
$ sudo LIBNFC_LOG_LEVEL=3 nfc-anticol 
info	libnfc.config	Unable to open file: /usr/local/etc/nfc/libnfc.conf
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
debug	libnfc.general	0 device(s) found using acr122_usb driver
debug	libnfc.driver.pn53x_usb	device found: Bus 020 Device 012
debug	libnfc.general	1 device(s) found using pn53x_usb driver
debug	libnfc.driver.pn53x_usb	3 element(s) have been decoded from "pn53x_usb:020:012"
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 02 fe d4 02 2a 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 06 fa d5 03 33 02 07 07 e5 00 
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 02 fe d4 02 2a 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 06 fa d5 03 33 02 07 07 e5 00 
debug	libnfc.chip.pn53x	SetParameters
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 03 fd d4 12 14 06 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 02 fe d5 13 18 00 
debug	libnfc.general	"SCM Micro / SCL3711-NFC&RW" (pn53x_usb:020:012) has been claimed.
debug	libnfc.general	set_property_bool NP_ACTIVATE_FIELD False
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 0c f4 d4 06 63 02 63 03 63 0d 63 38 63 3d b0 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 08 f8 d5 07 00 00 00 00 00 07 1d 00 
debug	libnfc.chip.pn53x	PN53X_REG_CIU_TxMode (Defines the transmission data rate and framing during transmission)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_RxMode (Defines the transmission data rate and framing during receiving)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_BitFraming (Adjustments for bit oriented frames)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 0b f5 d4 08 63 02 80 63 03 80 63 3d 00 b9 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 03 fd d5 09 00 22 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 04 fc d4 32 01 00 f9 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 02 fe d5 33 f8 00 
debug	libnfc.chip.pn53x	PN53X_SFR_P3 (Port 3 value)
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 04 fc d4 06 ff b0 77 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 04 fc d5 07 00 3e e6 00 
debug	libnfc.general	set_property_bool NP_ACTIVATE_FIELD True
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 04 fc d4 32 01 01 f8 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 02 fe d5 33 f8 00 
debug	libnfc.chip.pn53x	PN53X_SFR_P3 (Port 3 value)
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 04 fc d4 06 ff b0 77 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 04 fc d5 07 00 3e e6 00 
debug	libnfc.chip.pn53x	PN53X_SFR_P3 (Port 3 value)
[... 407 lines cut ...]
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	InRelease
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 03 fd d4 52 00 da 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 03 fd d5 53 00 d8 00 
debug	libnfc.general	set_property_bool NP_ACTIVATE_FIELD False
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 04 fc d4 32 01 00 f9 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 02 fe d5 33 f8 00 
debug	libnfc.chip.pn53x	PN53X_SFR_P3 (Port 3 value)
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 04 fc d4 06 ff b0 77 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 04 fc d5 07 00 3a ea 00 
debug	libnfc.chip.pn53x	PN53X_SFR_P3 (Port 3 value)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.pn53x_usb	TX: 00 00 ff 05 fb d4 08 ff b0 3e 37 00 
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.driver.pn53x_usb	RX: 00 00 ff 03 fd d5 09 00 22 00
```

doegox, 2020-08-01:
All seems normal to me, just that nfc-anticol didn't see any tag in the second and third runs.
Did you try with a genuine iso14443A tag ?