nfc-tools/libnfc #628: [Question] How to use card reader

bosgnoom, 2020-11-13:
Recently I've bought an NFC RFID reader, to be used with libnfc and some hobby projects. This instead of a pn_532 PCB, which is working fine. As you might guess, the NFC reader does not work on my system. I've compiled libnfc and installed successfully. However, the reader identifies as a "NSCCN wCopy smart reader", with USB ID 6869:1256.

> nfc-list uses libnfc libnfc-1.8.0-47-g7ebf9b9
No NFC device found.

Do you have any idea how to get this reader functioning? I've carefully opened the device. It shows an PN533 chip inside. Libnfc is compiled with pn533_usb support.

douardda, 2020-11-20:
I received this same device (was an error from the sender, anyway).
It indeed offers the USB ID 6869:1256 and advertise 2 USB interfaces, one mass storage and one HID:
```
Bus 001 Device 027: ID 6869:1256 NSCCN wCopy smart reader
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x6869 
  idProduct          0x1256 
  bcdDevice            0.00
  iManufacturer           1 NSCCN
  iProduct                2 wCopy smart reader
  iSerial                 3 A02011:55:16
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              480mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      29
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
[... 123 lines cut ...]
nov. 20 21:24:55 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 01 00 00 07 00
nov. 20 21:24:55 libra kernel: Dev sda: unable to read RDB block 0
nov. 20 21:24:55 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=476, sector_sz=512)
nov. 20 21:24:55 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 01 00
nov. 20 21:24:55 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=3548, sector_sz=512)
nov. 20 21:24:55 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 01 00 00 07 00
nov. 20 21:24:55 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=4060, sector_sz=512)
nov. 20 21:24:55 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
nov. 20 21:24:55 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=476, sector_sz=512)
nov. 20 21:24:55 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 18 00 00 01 00
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=3548, sector_sz=512)
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 19 00 00 07 00
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=476, sector_sz=512)
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 01 00
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=3548, sector_sz=512)
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 00 01 00 00 07 00
nov. 20 21:24:56 libra kernel:  sda: unable to read partition table
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] Attached SCSI removable disk
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=4060, sector_sz=512)
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 0f 80 00 00 08 00
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=476, sector_sz=512)
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 0f 80 00 00 01 00
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] Unaligned partial completion (resid=3548, sector_sz=512)
nov. 20 21:24:56 libra kernel: sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 0f 81 00 00 07 00
```

bosgnoom, 2020-11-26:
Yes, that looks familiar. 

By the way, when plugged into a windows system the mass storage device works. It contains a program, to be used with the card reader. Running **strings** on it shows a lot of references to libnfc, pn53x and mfoc...

In the mean time I've acquired a ACR122U-A9 device, which is running fine using my Fedora Linux system.

Krzysztof-Dziardziel, 2020-12-09:
Sadly, I have the same issue. I'm afraid that this device might be incompatible in some way.

neomilium, 2021-01-04:
This device is not supported according to README file.

closed 2021-01-04 by neomilium

bosgnoom, 2021-01-05:
OK, thank you for confirming. I did not see this specific device in the README, otherwise I wouldn't have created an issue.

Krzysztof-Dziardziel, 2021-01-05:
I haven't seen it either. Good to know that now. Thank You.

sfornengo, 2023-03-07:
Mine identifies as a "Winbond Electronics Corp. wCopy Smart Reader", with USB ID 0416:b030.
Did someone find a workaround ?